adafruit/Adafruit_TinyUSB_Arduino #1: Missing header file: Adafruit_TinyUSB_Core.h

bjc, 2019-05-13:
Git rev: 7b4814f6253dd06969a817adad2db80bcbac2226

Can't compile any of the examples due to missing header file.

```
Arduino: 1.8.9 (Linux), Board: "SparkFun Pro nRF52840 Mini, s140 6.1.1 r0, Level 0 (Release)"

In file included from /home/bjc/Arduino/libraries/Adafruit_TinyUSB_Arduino/src/Adafruit_TinyUSB.h:28,
                 from /home/bjc/Arduino/libraries/Adafruit_TinyUSB_Arduino/examples/HID/hid_composite/hid_composite.ino:4:
/home/bjc/Arduino/libraries/Adafruit_TinyUSB_Arduino/src/Adafruit_USBD_MSC.h:28:10: fatal error: Adafruit_TinyUSB_Core.h: No such file or directory
 #include "Adafruit_TinyUSB_Core.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board SparkFun Pro nRF52840 Mini.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

hathach, 2019-05-13:
please use the current master branch of the BSP  https://github.com/adafruit/Adafruit_nRF52_Arduino
The library support BSP isn't released yet.

closed 2019-05-13 by hathach

osimons, 2019-06-04:
This library is linked to from the PyGamer guide @ https://learn.adafruit.com/adafruit-pygamer/arduino-libraries and without the missing Core files it will not compile. Is it the guide or the library or perhaps some other location where the library should be found?

PTS93, 2019-06-04:
@osimons you need to install the latest Adafruit SAMD core, tinyusb is only supported since core version 1.5.

osimons, 2019-06-04:
Thanks. I get it working using latest Arduino IDE 1.8.9 with SAMD boards + libraries. I was however trying to recreate the tutorial & 'arcada_pygamer_test' example using latest PlatformIO (4.0.0.a18), but my approach may be flawed.

hathach, 2019-06-05:
did you select Tinyusb in the` menu -> Tools -> USB Stack`

osimons, 2019-06-05:
Yes, @hathach. All is fine in Arduino IDE.

I'm instead trying to use PlatformIO with library dependencies, but I'm unable to locate the TinyUSB_Core files in that build environment.

stringlapse, 2019-07-03:
Looks like the PIO Atmel SAM platform has Adafruit SAMD core 1.4.1 right now, I guess we'll just have to wait for it to be updated to 1.5+

ladyada, 2019-07-04:
yes we're doing a 3-repo update :)

CodeOhms, 2020-11-01:
Still have the exact same issue in Platformio 1.50.1 :rage: 

```
In file included from .pio/libdeps/bluepill_f103c8_128k/Adafruit TinyUSB Library/src/Adafruit_TinyUSB.h:28,
                 from src/main.cpp:12:
.pio/libdeps/bluepill_f103c8_128k/Adafruit TinyUSB Library/src/Adafruit_USBD_HID.h:28:10: fatal error: Adafruit_TinyUSB_Core.h: No such file or directory

*******************************************************************************
* Looking for Adafruit_TinyUSB_Core.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Adafruit_TinyUSB_Core.h"
* Web  > https://platformio.org/lib/search?query=header:Adafruit_TinyUSB_Core.h
*
*******************************************************************************

   28 | #include "Adafruit_TinyUSB_Core.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/bluepill_f103c8_128k/src/main.cpp.o] Error 1
In file included from .pio/libdeps/bluepill_f103c8_128k/Adafruit TinyUSB Library/src/Adafruit_TinyUSB.h:28,
                 from .pio/libdeps/bluepill_f103c8_128k/Adafruit TinyUSB Library/src/Adafruit_TinyUSB.cpp:25:
.pio/libdeps/bluepill_f103c8_128k/Adafruit TinyUSB Library/src/Adafruit_USBD_HID.h:28:10: fatal error: Adafruit_TinyUSB_Core.h: No such file or directory

*******************************************************************************
* Looking for Adafruit_TinyUSB_Core.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Adafruit_TinyUSB_Core.h"
* Web  > https://platformio.org/lib/search?query=header:Adafruit_TinyUSB_Core.h
*
*******************************************************************************

   28 | #include "Adafruit_TinyUSB_Core.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/bluepill_f103c8_128k/lib30b/Adafruit TinyUSB Library/Adafruit_TinyUSB.cpp.o] Error 1
```

PTS93, 2020-11-01:
@CodeOhms 
It seems you are using a STM32 based board which is not supported by this Arduino library yet. If you want to use TinyUSB with STM32 checkout the core project but keep in mind that this is bare C and does not come with Arduino support: https://github.com/hathach/tinyusb