adafruit/Adafruit_TinyUSB_Arduino #4: while(!Serial.available()); crashes with TinyUSB enabled sketch

cogliano, 2019-06-04:
I've added code to the msc_external_flash.ino example sketch to illustrate the issue. If you replace the loop routine in the sketch with this:

```
void loop()
{
  Serial.print("Press a key: ");
  while(!Serial.available()){}
  //while(!Serial.available()){delay(50);}
  int c = Serial.read();
  Serial.println("Got key " + String(c));
  delay(500);
}

```
The sketch will crash. However, if you instead use the other while loop with the delay(50), the sketch works. Perhaps an interrupt issue? The delay() technique can be used as a work-around .

ladyada, 2019-06-04:
yep for now you must put in a yield() or delay() to have it run the USB code since we handle MSC in 'userspace'!

hathach, 2019-06-05:
although it is advised to call delay/yield when blocking wait on usb event, I just added this PR to help with this particular issue. https://github.com/adafruit/ArduinoCore-samd/pull/122

It is worth the fix since this code appears often enough in user sketch and library
```
while(!Serial.available()) {}
```

hathach, 2019-06-11:
Should be fixed in the next samd core release (cdc is part of core)

closed 2019-06-11 by hathach