adafruit/Adafruit_TinyUSB_Arduino #6: HID + MSC

hathach, 2019-06-17:
It works on SAMD, but only if HID is initialized before MSD. @PaintYourDragon

ATMakersBill, 2019-07-05:
To be clear, do you mean that the Arduino Mouse and Keyboard functions work?  Or that you can define your own HID descriptor etc like in the examples?

I have been unable to use any builtin Mouse/Keyboard with TinyUSB at this point.

hathach, 2019-07-06:
@ATMakersBill all examples with hid work well now on ubuntu. Hid descriptor is changable for flexibility, the example use pretty standard descriptor for mouse and keyboard.

Maybe it is os issue, Which OS you are testing, I will give it a try.

ATMakersBill, 2019-07-06:
I'm sorry, I wasn't being clear.

Will this USB stack support the standard Arduino Mouse and Keyboard interfaces?  That is, will there be an option to use the examples that use 
#include <Mouse.h>
and have that logic work with the TinyUSB stack?

ATMakersBill, 2019-07-06:
If it IS supposed to work with Mouse.h etc, then I'm getting these errors:
```
In file included from C:\Users\billb\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.1\cores\arduino/USB/PluggableUSB.h:23:0,

                 from C:\Users\billb\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.1\libraries\HID/HID.h:24,

                 from C:\Program Files (x86)\Arduino\libraries\Mouse\src/Mouse.h:25,

                 from D:\Documents\ArduinoOld\WiiMouse\WiiMouse.ino:5:

C:\Users\billb\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.1\cores\arduino/USB/USBAPI.h:118:23: error: conflicting declaration 'USBDeviceClass USBDevice'

 extern USBDeviceClass USBDevice;

                       ^

In file included from C:\Users\billb\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.1\cores\arduino/Adafruit_TinyUSB_Core/Adafruit_TinyUSB_Core.h:35:0,

                 from C:\Users\billb\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.1\cores\arduino/Arduino.h:129,

                 from C:\Users\billb\AppData\Local\Temp\arduino_build_203410\sketch\WiiMouse.ino.cpp:1:

C:\Users\billb\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.1\cores\arduino/Adafruit_TinyUSB_Core/Adafruit_USBD_Device.h:66:29: error: 'USBDevice' has a previous declaration as 'Adafruit_USBD_Device USBDevice'

 extern Adafruit_USBD_Device USBDevice;

                             ^

In file included from C:\Users\billb\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.1\cores\arduino/USB/PluggableUSB.h:23:0,

                 from C:\Users\billb\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.1\libraries\HID/HID.h:24,

                 from C:\Program Files (x86)\Arduino\libraries\Mouse\src/Mouse.h:25,

                 from D:\Documents\ArduinoOld\WiiMouse\WiiMouse.ino:5:

C:\Users\billb\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.1\cores\arduino/USB/USBAPI.h:189:16: error: conflicting declaration 'Serial_ Serial'

 extern Serial_ Serial;

                ^

In file included from C:\Users\billb\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.1\cores\arduino/Adafruit_TinyUSB_Core/Adafruit_TinyUSB_Core.h:36:0,

                 from C:\Users\billb\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.1\cores\arduino/Arduino.h:129,

                 from C:\Users\billb\AppData\Local\Temp\arduino_build_203410\sketch\WiiMouse.ino.cpp:1:

C:\Users\billb\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.1\cores\arduino/Adafruit_TinyUSB_Core/Adafruit_USBD_CDC.h:56:26: error: 'Serial' has a previous declaration as 'Adafruit_USBD_CDC Serial'

 extern Adafruit_USBD_CDC Serial;

                          ^

In file included from C:\Program Files (x86)\Arduino\libraries\Mouse\src/Mouse.h:25:0,

                 from D:\Documents\ArduinoOld\WiiMouse\WiiMouse.ino:5:

C:\Users\billb\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.1\libraries\HID/HID.h:72:3: error: 'InterfaceDescriptor' does not name a type

   InterfaceDescriptor hid;

   ^

C:\Users\billb\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.1\libraries\HID/HID.h:74:3: error: 'EndpointDescriptor' does not name a type

   EndpointDescriptor  in;

   ^
```

ladyada, 2019-07-06:
these are totally different USB stacks, they are not cross compatible. there are HID examples in the library
[image]
please try those

hathach, 2019-07-07:
Hmm, maybe we should have adaptor to make it compatiblebwith existing Arduino API. What do you think @ladyada @ATMakersBill

ladyada, 2019-07-07:
maybe later, after everything is working well :)

ATMakersBill, 2019-07-07:
Ok: if the api is stable, there are folks who are no Ha who can work on higher layer abstractions. :-)

I was confused because there is a PluggableUSB api in Arduino and I thought this was using that to be a drop in replacement. Perhaps a note that it’s not the case would be good.

hathach, 2019-09-14:
Closed and reopen if it re-emerge

closed 2019-09-14 by hathach