adafruit/Adafruit_TinyUSB_Arduino #8: Conflicts compiling w/tinyusb selected.

ATMakersBill, 2019-07-04:
I'm getting the following error when building with Arduino 1.8.8... I've updated my boards: am I missing a step here?  I get no errors when using the Arduino USB stack, but when I switch I get conflicts.

I'm anxious to use the USB Filesystem with my Arduino projects (thanks!)

```
"C:\\Users\\billb\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -Os -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD "-D__SKETCH_NAME__=\"\"\"WiiMouse.ino\"\"\"" -DF_CPU=48000000L -DARDUINO=10808 -DARDUINO_TRINKET_M0 -DARDUINO_ARCH_SAMD -DCRYSTALLESS -DADAFRUIT_TRINKET_M0 -D__SAMD21E18A__ -DARM_MATH_CM0PLUS -DUSB_VID=0x239A -DUSB_PID=0x801E -DUSBCON "-DUSB_MANUFACTURER=\"Adafruit\"" "-DUSB_PRODUCT=\"Trinket M0\"" -DUSE_TINYUSB "-IC:\\Users\\billb\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.5.1\\cores\\arduino/Adafruit_TinyUSB_Core" "-IC:\\Users\\billb\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.5.1\\cores\\arduino/Adafruit_TinyUSB_Core/tinyusb/src" "-IC:\\Users\\billb\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS\\4.5.0/CMSIS/Include/" "-IC:\\Users\\billb\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS-Atmel\\1.2.0/CMSIS/Device/ATMEL/" "-IC:\\Users\\billb\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.5.1\\cores\\arduino" "-IC:\\Users\\billb\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.5.1\\variants\\trinket_m0" "-IC:\\Users\\billb\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.5.1\\libraries\\Wire" "-IC:\\Program Files (x86)\\Arduino\\libraries\\Mouse\\src" "-IC:\\Users\\billb\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.5.1\\libraries\\HID" "C:\\Users\\billb\\AppData\\Local\\Temp\\arduino_build_790950\\sketch\\WiiMouse.ino.cpp" -o "C:\\Users\\billb\\AppData\\Local\\Temp\\arduino_build_790950\\sketch\\WiiMouse.ino.cpp.o"
In file included from C:\Users\billb\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.1\cores\arduino/USB/PluggableUSB.h:23:0,

                 from C:\Users\billb\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.1\libraries\HID/HID.h:24,

                 from C:\Program Files (x86)\Arduino\libraries\Mouse\src/Mouse.h:25,

                 from D:\Documents\ArduinoOld\WiiMouse\WiiMouse.ino:5:

C:\Users\billb\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.1\cores\arduino/USB/USBAPI.h:118:23: error: conflicting declaration 'USBDeviceClass USBDevice'

 extern USBDeviceClass USBDevice;

                       ^

In file included from C:\Users\billb\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.1\cores\arduino/Adafruit_TinyUSB_Core/Adafruit_TinyUSB_Core.h:35:0,

                 from C:\Users\billb\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.1\cores\arduino/Arduino.h:129,

                 from C:\Users\billb\AppData\Local\Temp\arduino_build_790950\sketch\WiiMouse.ino.cpp:1:

C:\Users\billb\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.1\cores\arduino/Adafruit_TinyUSB_Core/Adafruit_USBD_Device.h:66:29: error: 'USBDevice' has a previous declaration as 'Adafruit_USBD_Device USBDevice'

 extern Adafruit_USBD_Device USBDevice;

                             ^

In file included from C:\Users\billb\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.1\cores\arduino/USB/PluggableUSB.h:23:0,

                 from C:\Users\billb\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.1\libraries\HID/HID.h:24,

                 from C:\Program Files (x86)\Arduino\libraries\Mouse\src/Mouse.h:25,

                 from D:\Documents\ArduinoOld\WiiMouse\WiiMouse.ino:5:

C:\Users\billb\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.1\cores\arduino/USB/USBAPI.h:189:16: error: conflicting declaration 'Serial_ Serial'

 extern Serial_ Serial;

                ^

In file included from C:\Users\billb\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.1\cores\arduino/Adafruit_TinyUSB_Core/Adafruit_TinyUSB_Core.h:36:0,

                 from C:\Users\billb\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.1\cores\arduino/Arduino.h:129,

                 from C:\Users\billb\AppData\Local\Temp\arduino_build_790950\sketch\WiiMouse.ino.cpp:1:

C:\Users\billb\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.1\cores\arduino/Adafruit_TinyUSB_Core/Adafruit_USBD_CDC.h:56:26: error: 'Serial' has a previous declaration as 'Adafruit_USBD_CDC Serial'

 extern Adafruit_USBD_CDC Serial;

                          ^

In file included from C:\Program Files (x86)\Arduino\libraries\Mouse\src/Mouse.h:25:0,

                 from D:\Documents\ArduinoOld\WiiMouse\WiiMouse.ino:5:

C:\Users\billb\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.1\libraries\HID/HID.h:72:3: error: 'InterfaceDescriptor' does not name a type

   InterfaceDescriptor hid;

   ^

C:\Users\billb\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.1\libraries\HID/HID.h:74:3: error: 'EndpointDescriptor' does not name a type

   EndpointDescriptor  in;

   ^
```

ATMakersBill, 2019-07-04:
I found a solution to this:
On the Preferences panel, if you uncheck "Agressively cache compiled core" it will rebuild and solve the conflict.

I would think there would be a way in the core or the USB stack to force this or actually do the dependency check, but I am ok disabling that cache every time I change stack.

We can close this, but we should probably not delete it in case anyone else has the issue.

[image]

hathach, 2019-07-04:
hmm, strange, I have that option checked as well but haven't got any issues. I am on 1.8.9, or maybe I don't switch stack very often. Thank you for your issue. I am sure others will find it useful.

Note: there is a breaking PR for Adafruit SPIFlash https://github.com/adafruit/Adafruit_SPIFlash/pull/13 It will change the way the SPIFlash is currnetly used. Basically use SdFat library for both external flash and sd card.

closed 2019-07-04 by hathach