adafruit/Adafruit_TinyUSB_Arduino #11: webusb

hathach, 2019-07-10:


PTS93, 2019-07-10:
Just to +1 this. WebUSB would be amazing. Existing projects are only in the proof of concept stage.

hathach, 2019-07-10:
@PTS93 I am new to webusb as well (reading specs now), if you have any reference or example that you think would be useful, please let me know.

PTS93, 2019-07-10:
The repo you posted is the only project I've seen in regard to Arduino (and board that have an Arduino core).
The uf2 bootloader also support WebUSB, not sure how well though
https://github.com/microsoft/uf2-samdx1

There is also something from the mbed world:
https://os.mbed.com/users/devanlai/code/USBDevice_STM32F103/file/e410de636542/WebUSBDevice/
https://github.com/devanlai/webdfu

commit 01897a8c (2019-08-02): Merge pull request #22 from adafruit/develop

fix #11 added support for webusb with webusb_serial example
closed 2019-08-02 by hathach