adafruit/Adafruit_TinyUSB_Arduino #13: Question:  Proper protocol for sleep on SAMD21

SRGDamia1, 2019-07-12:
When putting a FeatherM0 to sleep I've always followed the advice from the LowPower library and others to detach the usb first (```USBDevice.detach();```).  The TinyUSB library doesn't seem to have any equivalent functions.  How should the USB connection be shut down before putting the board into standby?

I'm not trying to use any of the drag-and-drop features of the TinyUSB library.  I'm working on another library that sleeps between logging sensor data.  But in  the newest SAMD Core the M0 defaults to using TinyUSB so my library won't compile any more.

ladyada, 2019-07-12:
the M0 doesn't default to tinyusb - select arduino in the usb stack dropdown menu

SRGDamia1, 2019-07-12:
Oh.  I'm using PlatformIO.  The USE_TINYUSB build flag is now set by default for the M0.  I'll raise an issue there to make sure it isn't the default.

But it would be nice to know what the proper protocol is.  Is sleep even possible (or sensible) when using TinyUSB?

ladyada, 2019-07-12:
agreed but we are trying to solve your problem right now. thanks :)

hathach, 2019-07-13:
Can you post a simplest sketch that reproduce the issue, a bit of comment explaining if it is not simple enough

hathach, 2020-01-01:
`USBDevice.detach()` is implemented in release 0.8.0 please try to see if that works.

closed 2020-01-01 by hathach