adafruit/Adafruit_TinyUSB_Arduino #14: won't compile on Feather M0

deladriere, 2019-07-17:


hathach, 2019-07-17:
please update this library to latest 0.5.0 then try again

closed 2019-07-19 by hathach

deladriere, 2019-07-20:
Ok thanks, It works in the Arduino IDE

deladriere, 2019-07-20:
@hathach 
It still doesn't compile with the Platformio IDE
Here is my platformio.ini
```
[env:adafruit_feather_m0]
platform = atmelsam
board = adafruit_feather_m0
framework = arduino
lib_deps =
    Adafruit TinyUSB Library
    MIDI Library
```

and the error 

```
dependency Graph
|-- <Adafruit TinyUSB Library> 0.5.0
|-- <MIDI Library> 4.3.1
Compiling .pio/build/adafruit_feather_m0/src/main.cpp.o
Compiling .pio/build/adafruit_feather_m0/lib1e2/Adafruit TinyUSB Library_ID6465/Adafruit_TinyUSB.cpp.o
Compiling .pio/build/adafruit_feather_m0/lib1e2/Adafruit TinyUSB Library_ID6465/Adafruit_USBD_HID.cpp.o
Compiling .pio/build/adafruit_feather_m0/lib1e2/Adafruit TinyUSB Library_ID6465/Adafruit_USBD_MIDI.cpp.o
src/main.cpp:23:20: error: cannot declare variable 'usb_midi' to be of abstract type 'Adafruit_USBD_MIDI'
 Adafruit_USBD_MIDI usb_midi;
                    ^~~~~~~~
```

hathach, 2019-07-20:
Please open an issue in the platformio, it is only a matter of update.

deladriere, 2019-07-20:
@hathach but they have the <Adafruit TinyUSB Library> 0.5.0 as you can see on the dependency Graph. what else do they need to update?

hathach, 2019-07-22:
I haven't used platformio, I have no idea about what to suggest. You better post this question to platformio for them to help.