adafruit/Adafruit_TinyUSB_Arduino #16: cannot declare variable 'usb_msc' to be of abstract type 'Adafruit_USBD_MSC'

Pedrinbeep, 2019-07-17:
Hi, I'm get this error on "sd" example. I just update the library, the first version (abobe 1 month ago) works. 

Also, the last version, only works if I open Serial on Arduino IDE and take 1 minute to appear the drive with the SD files on it. What I'm doing wrong?

Thanks you

hathach, 2019-07-17:
Can you provide more information on your setup and the exact error message of build process

**Set up (please complete the following information)**
 - Your Board and BSP version
 - TinyUSB library version 
 - Your OS (mac/windows/linux) and its version
 - Serial debug log if any

Pedrinbeep, 2019-07-17:
Okey the problem just solved! The board was Adafruit Grand Central M4, and right now the Adafruit SAMD library just updated and the error don't appear any more. 

About the time to appear the SD drive. The example sketch I using is this: 

`/*********************************************************************
 Adafruit invests time and resources providing this open source code,
 please support Adafruit and open-source hardware by purchasing
 products from Adafruit!

 MIT license, check LICENSE for more information
 Copyright (c) 2019 Ha Thach for Adafruit Industries
 All text above, and the splash screen below must be included in
 any redistribution
*********************************************************************/

/* This example expose SD card as mass storage using
 * SdFat Library
 */

#include "SPI.h"
#include "SdFat.h"
#include "Adafruit_TinyUSB.h"

const int chipSelect = SDCARD_SS_PIN;

// File system on SD Card
SdFat sd;

SdFile root;
SdFile file;

// USB Mass Storage object
Adafruit_USBD_MSC usb_msc;

// Set to true when PC write to flash
bool changed;

// the setup function runs once when you press reset or power the board
void setup()
{
  pinMode(LED_BUILTIN, OUTPUT);

  // Set disk vendor id, product id and revision with string up to 8, 16, 4 characters respectively
  usb_msc.setID("Adafruit", "SD Card", "1.0");

  // Set read write callback
  usb_msc.setReadWriteCallback(msc_read_cb, msc_write_cb, msc_flush_cb);

  // Still initialize MSC but tell usb stack that MSC is not ready to read/write
  // If we don't initialize, board will be enumerated as CDC only
  usb_msc.setUnitReady(false);
  usb_msc.begin();

  Serial.begin(115200);
  while ( !Serial ) delay(10);   // wait for native usb

  Serial.println("Adafruit TinyUSB Mass Storage SD Card example");

  Serial.print("\nInitializing SD card ... ");
  Serial.print("CS = "); Serial.println(chipSelect);

  if ( !sd.begin(chipSelect, SD_SCK_MHZ(50)) )
  {
    Serial.println("initialization failed. Things to check:");
    Serial.println("* is a card inserted?");
    Serial.println("* is your wiring correct?");
    Serial.println("* did you change the chipSelect pin to match your shield or module?");
    while (1) delay(1);
  }

[... 114 lines cut ...]
14:28:50.662 ->          0 .Trashes/
14:28:50.662 ->         33 SHOW.TXT
14:28:50.662 -> 
14:28:51.685 -> SD contents:
14:28:51.685 ->          0 .Spotlight-V100/
14:28:51.685 ->          0 .fseventsd/
14:28:51.685 ->         33 SETTINGS.TXT
14:28:51.685 ->          0 .Trashes/
14:28:51.685 ->         33 SHOW.TXT
14:28:51.685 -> 
14:28:52.674 -> SD contents:
14:28:52.674 ->          0 .Spotlight-V100/
14:28:52.674 ->          0 .fseventsd/
14:28:52.674 ->         33 SETTINGS.TXT
14:28:52.674 ->          0 .Trashes/
14:28:52.674 ->         33 SHOW.TXT
14:28:52.674 -> 
```

**Set up info:**
Board: Adafruit Grand Central M4
TinyUSB library version: 0.5.0 (The last one)
OS: macOS 10.14.5

Thank you for the fast answer

hathach, 2019-07-17:
Example code wait for Serial connect to run, modify it of you dont want to wait

closed 2019-07-17 by hathach