adafruit/Adafruit_TinyUSB_Arduino #18: [Bug]

PTS93, 2019-07-22:
The HID in/out example is not working, there is no debug messages printed to the console and the hid raw device is not accessible through node-hid with the provided example code.

**Setup**
 - Adafruit SAMD 1.5.2
 - TinyUSB 0.5.0
 - Ubuntu 19.04 (same behaviour on 18.04)

**To Reproduce**
Upload generic in out example to SAMD21 board.

**Expected behavior**
Should print "Adafruit TinyUSB HID Generic In Out example" to serial console and node.js example should be able to connect and sent data back and forth.

PTS93, 2019-07-22:
The hid_mouse example is working.
Maybe related, when I run
`lsusb -v -d 239a:8018`
I posts:
`Bus 003 Device 040: ID 239a:8018  
Couldn't open device, some information will be missing`

It still posts all the information though (running the generic in/out example)
```
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x239a 
  idProduct          0x8018 
  bcdDevice            1.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x006b
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass          2 Communications
      bFunctionSubClass       2 Abstract (modem)
      bFunctionProtocol       1 AT-commands (v.25ter)
      iFunction               0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      1 AT-commands (v.25ter)
      iInterface              0 
      CDC Header:
        bcdCDC               1.20
      CDC Call Management:
        bmCapabilities       0x00
        bDataInterface          1
      CDC ACM:
        bmCapabilities       0x02
          line coding and serial state
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              16
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      32
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               2
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               2

```

PTS93, 2019-07-22:
Ok this is a permission issue. I didn't notice right away because when running the example with sudo it was switching to a different node.js version for the root user (I'm using nvm to manage node versions) which was not compatible to the compile node-hid for the older node.js version from which I was installing it from.
This threw an unrelated error.
Now that I temporarily set the permissions to 666, for the the hidraw device my CPY express was mapped to, it started working just fine.

closed 2019-07-22 by PTS93

dhalbert, 2019-07-22:
What is the group for the hidraw device? You can just add yourself to that group, or, if you want to be more complicated, you can make up some udev rules. But maybe you know that already.
https://learn.adafruit.com/adafruit-arduino-ide-setup/linux-setup#udev-rules

PTS93, 2019-07-22:
Yes, I now made a udev rule for this. By default Ubuntu is only allowing `root` access to hidraw. Which goes for any OS I guess.
This is my rule if anyone stumbles over this: `KERNEL=="hidraw*", SUBSYSTEM=="hidraw", ATTRS{idVendor}=="239a", MODE:="0660", GROUP="plugdev"`

This allows non root users in the plugdev group to read and write from Adafruit boards. You need to change this to a different vendor ID for other board vendors.
NEVER EVER leave out the vendorId part, otherwise you allow pretty much any program to read out your mouse and keyboard, e.g. doors are open for keyloggers etc.