adafruit/Adafruit_TinyUSB_Arduino #21: webusb example - RGB LED

ladyada, 2019-07-31:
please have an example for 3 x PWM output - so we can use https://webusb.github.io/arduino/demos/rgb/

PTS93, 2019-07-31:
Kind of related, where is the arduino sketch for the existing webusb example?

hathach, 2019-07-31:
@PTS93 after these 2 PRs 
https://github.com/adafruit/Adafruit_nRF52_Arduino/pull/313
https://github.com/adafruit/Adafruit_TinyUSB_Arduino/pull/22

starting with nrf first, I will update the samd core later on.

hathach, 2019-07-31:
> please have an example for 3 x PWM output - so we can use https://webusb.github.io/arduino/demos/rgb/

I think it is better to use on-board neopixel which is on most of our feather board. 3 PWM work as well, but it is secondary via compiler switch.

ladyada, 2019-07-31:
OK yes neopixel is good!

hathach, 2019-08-02:
@PTS93 it is merged to https://github.com/adafruit/Adafruit_TinyUSB_Arduino/tree/master/examples/WebUSB/webusb_serial

PTS93, 2019-08-02:
Awesome! I wanted to make a more elaborate example with the Circuit Playground Express unless there are already plans for something like that?

hathach, 2019-08-02:
@PTS93 feel free to do so, I push an PR for webusb-rgb to control on-board neopixel already, you may  want to check that out. My javascript and html already exceed its limit with these 2 examples :D
https://github.com/adafruit/Adafruit_TinyUSB_Arduino/pull/23

closed 2019-08-02 by hathach

PTS93, 2019-08-02:
That's helpful, thank you!
My web frontend skills are a bit rusty but I should be able to make something nice.
Do I understand correctly that with webusb you just send buffers back and forth like with raw hid?
That serial.js doesn't look like its doing anything that would be specific to classical UART.

hathach, 2019-08-05:
@PTS93 webUSB specs actually doesn't say that, it is more like an optional extension. However, this Arduino library did implement the webusb as vendor class driver, and indeed as an subclass of Stream much like Serial for user convenience.

commit 1a2b141b (2019-08-02): Merge pull request #23 from adafruit/develop

resolve #21 added webusb RGB example