adafruit/Adafruit_TinyUSB_Arduino #25: [Feature] Firmata on webusb example

hathach, 2019-08-06:
- https://github.com/firmata/firmata.js
- https://github.com/firmata/firmata.js/tree/master/packages/firmata.js

PTS93, 2019-08-07:
I'm currently focusing on getting osc to work but after that I could also help with firmata if help is still needed.

PTS93, 2019-08-07:
Though for all these kinds of libraries it would pay off to port node-serialport to the web. If thats done everything else would just work out of the box.
https://github.com/serialport/node-serialport

They have a binding layer where one could start making a binding for webusb.

ladyada, 2019-08-07:
@PTS93 hiya yep thach needs to look at some other code, so if you want to try firmata implementation that would be cool! there's probably existing code to use :)

hathach, 2019-08-07:
@PTS93 Yes, please. I was actually thinking of you when creating this issue. It will take me forever to implement this thanks to my "copy and paste" javascript skill :laughing:

PTS93, 2019-08-07:
A 'dirty' implementation would probably be fairly straight forward. E.g. clone firmata.js and replace the node-serialport portion.

A proper port that would enable other frameworks (possibly johnny-five) would be much more involved node-serialport is very large and very node.js specific, the build process would likely have to change to accommodate a browser build, right now its all C++ bindings.

Another possible route would be to implement a browser only node-serialport compatible project that could potentially be merged into node-serialport at a later point.

hathach, 2019-08-08:
@PTS93 I am OK with anything that works :smile:

PTS93, 2019-09-09:
Just wanted to give an update, I was quite busy with other stuff (mostly cccamp) the past weeks. I haven't forgotten about it but are currently not actively working on this. Maybe in the coming weeks.

hathach, 2019-09-10:
@PTS93 no hurry, I am actually buried in pile of work anyway :smile:

closed 2023-03-24 by hathach