adafruit/Adafruit_TinyUSB_Arduino #28: Static escriptor buffer size for large USB configurations?

kaysievers, 2019-08-26:
Enabling 16 virtual wires/jacks at a MIDI device grows the USB configuration/MIDI descriptor to 643 bytes. :) The static buffer in Adafruit_USBD_Device::_desc_cfg[256] cannot carry this.

What could be done about this? Would it be possible to just compose the buffer content on-the-fly/stack and return it from there, or does it need to be static/stay around?

kaysievers, 2019-08-27:
The  PR is https://github.com/adafruit/ArduinoCore-samd/pull/173

hathach, 2019-08-27:
the descriptor needs to live long enough until the GET_CONFIGURATION transfer is complete, which happens ~ 1ms after the `tud_descriptor_configuration_cb()` is called. So yes, it need to stay long enough.

I wouldn't increase the static size since 95% of sketch won't need it, also don't want to use malloc since I want to keep it as static and portable as possible. Since it isn't commonly use, it makes sense to add an API() called `setDescriptorBuffer(uint8_t* buf, uint32_t buflen); to optionally increase this buffer.

closed 2019-09-01 by kaysievers