adafruit/Adafruit_TinyUSB_Arduino #33: usb_hid.end()

emtark, 2019-10-06:
As requested by hathach on the forums - https://forums.adafruit.com/viewtopic.php?f=25&t=154636#p770791

Are we able to implement a end/detach function? 
Similar to existing request https://github.com/adafruit/Adafruit_TinyUSB_Arduino/issues/13

I'm using SAMD51 and unable to resume USB functions after returning from sleep/standby. I believe closing the connection, then sleeping, then reconnecting after waking up is the go.

Cheers,
emtark.

ladyada, 2019-10-06:
@hathach see this library https://github.com/adafruit/Adafruit_SleepyDog

hathach, 2019-10-09:
> As requested by hathach on the forums - https://forums.adafruit.com/viewtopic.php?f=25&t=154636#p770791
>
> Are we able to implement a end/detach function?
> Similar to existing request #13
>
> I'm using [...]

Can you provide your simplest sketch that could reproduce the issue with a bit of explanation on before and after behavior of sleep/standby statement within the sketch.

hathach, 2020-01-30:
@emtark USBDevice `detach() & attach()` can be used to physically disconnect/reconnect usb. Please try it out to see it works for you.
https://github.com/adafruit/Adafruit_TinyUSB_ArduinoCore/blob/master/Adafruit_USBD_Device.h#L84

emtark, 2020-02-03:
Works perfectly - thanks :)
On another note, (if required I can move to a new thread)
`
void hid_report_callback(uint8_t report_id, hid_report_type_t report_type, uint8_t const* buffer, uint16_t bufsize)
{`
This function fires on my PC when the USB keyboard example is ready to send data -but never called on Android. Is there another method I can use to ensure the Android OS is ready to receive USB HID keyboard data?

hathach, 2020-02-08:
Close since it solves op issue pleas. For hid callback it fires when there is related event. Please check out its contents for reported events.

closed 2020-02-08 by hathach