adafruit/Adafruit_TinyUSB_Arduino #35: Crashing on SAMD

PTS93, 2019-12-09:
Using SAMD Core 1.5.7 and TinyUSB_Arduino 0.7.1 I can't get any of the examples I tried to work, the Serial port never comes back after uploading and Windows reports a malfunctioning USB device.
Tested on SAMD21 and SAMD51.

With just the blink example uploaded but TinyUSB selected it is still working.

I tested generic HID and Keyboard HID. Both crash. I also tried MSC Ramdisk, which shows up after a while and the serial port also comes back but then disappears again after a bit.

hathach, 2019-12-10:
BSP and library are probably out of sync, thanks for reporting. I will check it out.

FAksel, 2019-12-19:
Hi there, also trying to get tinyusb MSC to work after updating the core. 
Same issue with the serial port not coming back after uploading. This was solved by updating the samd core with adadruit's latest commit. However it is now the SD card not initializing. 
The example msc_sd works fine but msc_sdfat doesnt: the cards fails to initialize.

hathach, 2019-12-20:
@FAksel thanks for confirmation, the core and library is out of sync. For your issue, SD card may not be reliable, can you verify that msc ramdisk and/or msc external flash are working ?

FAksel, 2019-12-20:
msc ramdisk works ok. Don't have external flash to test. 

It is more a spi problem though as none of the SdFat examples work.

I could solve the problem by reverting the changes in SPI.h brought by this commit:
https://github.com/adafruit/ArduinoCore-samd/commit/516cec5a9b927b0890485d76b49508442a85c2cf

hathach, 2019-12-20:
@FAksel thanks, if you think it is spi problem , please open a separated issue for it in samd core so that others could look at it.

FAksel, 2019-12-20:
Commented [PR 186](https://github.com/adafruit/ArduinoCore-samd/pull/186) with the problem.
Thank you.

PTS93, 2019-12-20:
Any eta when they are aligned again? Would like to work with a working "official" combination for an upcoming project.

hathach, 2019-12-20:
soon enough, we are doing a re-org 
https://github.com/adafruit/ArduinoCore-samd/pull/190

hathach, 2020-01-01:
TInyUSB in BSP and Library are released and should be in synced, please try again to if that work for you.

PTS93, 2020-01-01:
Thanks! Seems to be working again with current SAMD and TinyUSB from the Library/Board Manager

closed 2020-01-01 by PTS93