adafruit/Adafruit_TinyUSB_Arduino #37: [Bug] WebUSB device fails to enumerate on Windows 10

kaysievers, 2019-12-20:
Since the restructured Core repositories, WebUSB devices fail to enumerate on Windows. I dumped the descriptor, but cannot see any obvious difference.

Everything appears to work fine on Linux.

I tried the ```webusb_rgb.ino``` example with the same result. See the attached screenshot:

[image]

kaysievers, 2019-12-21:
Reverting this commit, makes it work again:

https://github.com/versioduo/Adafruit_TinyUSB_ArduinoCore/commit/87d4275798cea11c880a45115116cd7cc799d2c8

hathach, 2019-12-23:
Thanks for the filing the issue, I am investigating, will try to fix it asap :)

hathach, 2019-12-24:
@kaysievers I push and update to the Core repo, it uses an WIP code that help to improve samd port, please git a git pull to see if that fixes this issue.
https://github.com/adafruit/Adafruit_TinyUSB_ArduinoCore

kaysievers, 2019-12-24:
It does not help. As soon as I add WebUSB descriptor, the device does no longer work.

It looks like a "random" corruption in the descriptor memory. I get a different behavior when I connect the device many times.

Here is a kernel log from Linux, you can see corrupted vendor/product strings (product here):
```
[59486.636718] usb 1-3: New USB device found, idVendor=6666, idProduct=d9b0, bcdDevice= 1.00
[59486.636723] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[59486.636726] usb 1-3: Product: Љ
[59486.636728] usb 1-3: Manufacturer: V2
[59486.636731] usb 1-3: SerialNumber: B36925245333395336202020FF094115
```

Here the serial number string is the same as the vendor string, the next connect looks ok:
```
[59552.785429] usb 1-3: new full-speed USB device number 43 using xhci_hcd
[59552.913854] usb 1-3: New USB device found, idVendor=6666, idProduct=d9b0, bcdDevice= 1.00
[59552.913859] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[59552.913863] usb 1-3: Product: pad blue
[59552.913865] usb 1-3: Manufacturer: V2
[59552.913868] usb 1-3: SerialNumber: V2
[59552.919419] cdc_acm 1-3:1.0: ttyACM0: USB ACM device
[59553.473370] usb 1-3: USB disconnect, device number 43
[59554.233416] usb 1-3: new full-speed USB device number 44 using xhci_hcd
[59554.362255] usb 1-3: New USB device found, idVendor=6666, idProduct=d9b0, bcdDevice= 1.00
[59554.362264] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[59554.362268] usb 1-3: Product: pad blue
[59554.362271] usb 1-3: Manufacturer: V2
[59554.362273] usb 1-3: SerialNumber: B36925245333395336202020FF094115
```

With every 15th connect, of the device fails to enumerate at all:
```
[59556.769356] usb 1-3: new full-speed USB device number 45 using xhci_hcd
[59556.897023] usb 1-3: unable to read config index 0 descriptor/all
[59556.897028] usb 1-3: can't read configurations, error -32
```

hathach, 2019-12-24:
@kaysievers sorry, it is still on my personal fork here 
https://github.com/hathach/ArduinoCore-samd/tree/develop

Please give a try again,I also fixed another racing issue with samd51 + QSPI as well. It should really works now. Else please let me know whic board you are testing and your windows build version (in **about** )

kaysievers, 2019-12-24:
> Please give a try again,

It does not work. Same issues.

I already tried the latest version in the report above. I do not rely on ArduinoCore-samd's submodules to be up-to-date, I did that myself already before the last test. All repos are on the current 'master' branch's version.

Adding WebUSB to the descriptor still corrupts the USB descriptor here. I'm running it on a SAMD51 custom board.

Im running:
ArduinoCore-samd 1941e1

with the updated submodule:
Adafruit_TinyUSB_ArduinoCore a90b6b

The library is at:
Adafruit_TinyUSB_Arduino 1284cc

> your windows build version

The same happens on Linux, it is not related to the operating system or the IDE.

hathach, 2019-12-24:
It happens once and is fixed on my machine with the update. Can you use one of the adafruit samd51 board just to make sure we are testing the same hw. It is hardware to cross board check as it is since I couldn’t reproduce the issue anymore.

kaysievers, 2019-12-24:
Tested it all again.

If I revert these two commits:
https://github.com/adafruit/Adafruit_TinyUSB_ArduinoCore/commit/a90b6b651c4777977b727a8f58d08ac1caf77437
https://github.com/adafruit/Adafruit_TinyUSB_ArduinoCore/commit/87d4275798cea11c880a45115116cd7cc799d2c8

everything appears to work fine,

kaysievers, 2019-12-24:
It appears that the issue is not related to WebUSB. I can see the same corrupted descriptor data without it, but it is far less likely.

I need to press the reset button approximately 100 times to see the issue. With WebUSB added, it happens approximately every 10th time, maybe just because it is so much larger.

With the above mentioned commits, all is stable, and I cannot trigger the descriptor data corruption at all.

hathach, 2019-12-24:
Thanks for testing, if possible please use one of the adafruit dev board. They seem to be stable enough for me. The hw issue can be related to pcb routing or such as well

kaysievers, 2019-12-24:
> The hw issue can be related to pcb routing or such as well

Well, it looks more like a software issue if reverting your recent changes makes it work again :)

hathach, 2019-12-25:
Hmm, tested again with Feather M4, I couldn't reproduce the issue. Can you try to set this to `CFG_TUSB_DEBUG` to 1, and hook up an FTDI to TX (Serial1) to see if the stack output any extra information. 
https://github.com/adafruit/ArduinoCore-samd/blob/master/cores/arduino/TinyUSB/tusb_config.h#L49

hathach, 2019-12-25:
@kaysievers please pull from core repo and try again
https://github.com/adafruit/Adafruit_TinyUSB_ArduinoCore/commit/8096c8a8f8aad8458ff5762c990e10ae9dad1a1b

improve the racing condition with setup/control hanlding.

kaysievers, 2019-12-25:
I pressed the reset button like 200 times now, and did not see an single failure. It appears to work all fine. I'll try the other setups later if they work too. Thanks a lot!

hathach, 2019-12-25:
> I pressed the reset button like 200 times now, and did not see an single failure. It appears to work all fine. I'll try the other setups later if they work too. Thanks a lot!

I am worrying for your reset button now :D. Thanks for testing it out and confirmation, this is a typical racing condition. The bug is there but it requires right timing between host requests to occur.

closed 2019-12-25 by hathach