adafruit/Adafruit_TinyUSB_Arduino #43: [Bug]0.8.1 undefined reference to `Adafruit_USBD_MIDI::Adafruit_USBD_MIDI()

deladriere, 2020-02-08:
I receive this error while compiling with Platformio
I was working fine with the 0.6.0 version
Here is my platformio ini

[env:adafruit_feather_m0]
platform = atmelsam
board = adafruit_feather_m0
build_flags = -D USE_TINYUSB
framework = arduino
lib_deps =
    Adafruit TinyUSB Library@0.8.1
    MIDI Library@4.3.1
    SSD1306Ascii@1.2.4
monitor_speed = 115200

deladriere, 2020-02-08:
@ivankravets can you check this too ?

hathach, 2020-02-08:
Can you confirm this issue on Arduino with latest adafruit samd bsp ?

deladriere, 2020-02-08:
@hathach I am running this with  platformio and it is still on the Adafruit SAMD Adafruit Core  1.5.3
could this be the reason?

hathach, 2020-02-08:
Close since the issue is not reproducible on Arduino IDE, and not bug of this repo

closed 2020-02-08 by hathach