adafruit/Adafruit_TinyUSB_Arduino #45: [Feature] There is a begin() method, but no end() method. Should there be one?

akamensky, 2020-02-29:
**Is your feature request related to a problem? Please describe.**
I'd like to de-init/de-activate USB when it is disconnected and make sure I does not spend CPU cycles on it wasting precious battery. At the moment I only see a `begin()` method and 0 documentation. Should there be an `end()` method as it is often implemented in other libraries?

**Describe the solution you'd like**
Either a documentation on how to disable interface or a clear `end()` method.

**Describe alternatives you've considered**
Don't see any alternative really.

PTS93, 2020-02-29:
See https://github.com/adafruit/Adafruit_TinyUSB_ArduinoCore/blob/master/Adafruit_USBD_Device.h#L84

That might help you. Though if depend on physical disconnects by the user already and don't need to do that programmatically you shouldn't need to do much more. You could also delete your USB class object if you want to save on memory too.
Completely disabling USB is not possible unless you are ok with loosing the ability to detect being plugged into a USB port or have a functioning serial bootloader and instead rely on the double reset method to load new firmware.

akamensky, 2020-02-29:
@PTS93 I guess I was not clear enough. The device will be plugged back eventually. But when that happens I'd rather prefer to re-init the connection again than rely that everything is in correct state already. Having `.end()` would give a clear option to de-init the interface (without physically disabling it) and having to re-init at again when device is connected again.

Not even sure if that would be needed, hence the original question whether this is needed.

For CPU cycles -- I can do this programmatically by detecting USB presence (USB 5V -> resistor -> GPIO as a crude way of doing it). But not sure if connection setup would be fully preserved and hence nothing else needs to be done, or I better call same setup steps and then `.begin()` again? Or would I need to clear something and then do my init?

With the total lack of documentation those items are not clear w/o digging into the code.

PTS93, 2020-02-29:
In that you don't need to do anything. TinyUSB takes care of that as much as possible.
Below is the main() that gets compiled when you select the TinyUSB core.
If there is no connection established all that happens is to check for a connection. All you are wasting is RAM, if that is no concern that there is nothing to do.

All these lower level things are defined in https://github.com/adafruit/Adafruit_TinyUSB_ArduinoCore and subsequently https://github.com/hathach/tinyusb/tree/master/src
Some device specific things are handled in the Arduino core itself. So for SAMD that would be https://github.com/adafruit/ArduinoCore-samd/tree/master/cores/arduino/TinyUSB

```
#define ARDUINO_MAIN
#include "Arduino.h"

// Weak empty variant initialization function.
// May be redefined by variant files.
void initVariant() __attribute__((weak));
void initVariant() { }

// Initialize C library
extern "C" void __libc_init_array(void);

/*
 * \brief Main entry point of Arduino application
 */
int main( void )
{
  init();

  __libc_init_array();

  initVariant();

  delay(1);

#if defined(USE_TINYUSB)
  Adafruit_TinyUSB_Core_init();
#elif defined(USBCON)
  USBDevice.init();
  USBDevice.attach();
#endif

  setup();

  for (;;)
  {
    loop();
    yield(); // yield run usb background task

    if (serialEventRun) serialEventRun();
  }

  return 0;
}

#if defined(USE_TINYUSB)

// run TinyUSB background task when yield()
extern  "C" void yield(void)
{
  tud_task();
  tud_cdc_write_flush();
}

#endif

```

closed 2021-05-25 by hathach