adafruit/Adafruit_TinyUSB_Arduino #47: [Bug] Weird dependency graph that causes compilation failures

akamensky, 2020-03-28:
**Describe the bug**
I am working on a project that uses this library from PlatformIO (VSCode), it fails trying to compile a module that I am not even remotely planning on using. Dependency graph is based on `depends` in `library.properties` and in case of this library it looks like this:

```
Scanning dependencies...
Dependency Graph
|-- <usb_hid>
|   |-- <Adafruit TinyUSB Library> 0.8.1
|   |   |-- <Adafruit SPIFlash> 3.1.4
|   |   |   |-- <SPI> 1.0
|   |   |   |-- <SdFat - Adafruit Fork> 1.2.3
|   |   |   |   |-- <SPI> 1.0
|   |   |-- <MIDI Library> 4.3.1
|   |   |-- <Adafruit seesaw Library> 1.3.0
|   |   |   |-- <Adafruit ST7735 and ST7789 Library> 1.5.15
|   |   |   |   |-- <Adafruit GFX Library> 1.7.5
|   |   |   |   |   |-- <Adafruit ILI9341> 1.5.4
|   |   |   |   |   |   |-- <Adafruit STMPE610> 1.1.0
|   |   |   |   |   |   |   |-- <SPI> 1.0
|   |   |   |   |   |   |-- <Adafruit TouchScreen> 1.0.4
|   |   |   |   |   |   |-- <SPI> 1.0
|   |   |   |   |   |-- <SPI> 1.0
|   |   |   |   |-- <SD>
|   |   |   |   |   |-- <SPI> 1.0
|   |   |   |   |-- <SPI> 1.0
|   |   |-- <Adafruit NeoPixel> 1.3.5
|   |   |-- <SdFat - Adafruit Fork> 1.2.3
|   |   |   |-- <SPI> 1.0
|   |   |-- <SD>
|   |   |   |-- <SPI> 1.0
|   |   |-- <Adafruit Circuit Playground> 1.10.4
|   |   |   |-- <SPI> 1.0
|-- <usb_sense>
```
The compilation failure is:
```
.platformio/lib/Adafruit STMPE610_ID377/Adafruit_STMPE610.cpp:32:10: fatal error: Wire.h: No such file or directory
```
For some unknown to me reason this library depends on STMPE610. Which in turn wants Wire.h, which is in turn not available (missing dependency?).

FWIW, it used to compile just fine until some (automated) library updates happened and now I can no longer compile, but again, not sure why would this library come anywhere close to needing  touchscreen, circuit playground and other weird dependencies that I see in deps graph.

**Set up (please complete the following information)**
 - 0.8.1
 - macos, but dont see how that would relate

**To Reproduce**
Steps to reproduce the behavior:
1. Use this library in a project 
2. Try to build a project

**Expected behavior**
Compilation succeeds

akamensky, 2020-03-28:
To fix compilation issue I needed to go and manually delete all those _pseudo-dependencies_, after which the project successful compiles.

I don't think those should be there then if they are not actually needed if this library works without them.

hathach, 2020-03-28:
yes, those libraries aren't needed by the core of this library, but it is included on purpose to compile the built-in examples. CircuitPlayground Express is required to run with one of the midi example on CPX board. CPX can in turn include other sensor dependency. We have no problem with Arduino IDE so far, eventually PlatformIO will sort this out.

closed 2020-03-28 by hathach