adafruit/Adafruit_TinyUSB_Arduino #48: github warning: potential security vulnerability in one of your dependencies.

hathach, 2020-03-30:
**Describe the bug**

Github bot detect some issue with minimist version older than 1.2.2 . However this is alien language to me, @PTS93 Do you have a minute to check this out :) 

[image]

[image]

todbot, 2020-03-30:
I would vote for removing `package-lock.json` from that directory and relying on `package.json`.  `package.json` shows what you intend, while `package-lock.json` is a snapshot of the entire state of the node world to allow an exact recreation. 

If you are curious, `minimist` is a dependency of a dependency in `node-hid`, a dependency every Node package with native code has.  Until that dependency gets updated (and its parent and finally `node-hid`), you'll see this message from the github bot.  When that dependency does get updated, I'll update `node-hid`.

hathach, 2020-03-30:
@todbot thank you for proposing your solution, to be honest, I have no idea about this. If you could submit an PR, I will be more than happy to test it out :)

PTS93, 2020-03-30:
Agreed package-lock.json is probably not important for a simple example like this. Usually its encouraged to commit it so other always work against the same dependency versions as you but that also means that someone has to regularly maintain it and I think thats a bit much for a hello world example.

hathach, 2020-03-30:
Thank you @PTS93 and @todbot for a quick response and solution, the warning message disappear now :+1: :+1:  . 

Fixed by #49

closed 2020-03-30 by hathach