adafruit/Adafruit_TinyUSB_Arduino #50: [Feature] keyboardPress method only supports ASCII characters

cyborg5, 2020-04-05:
**Is your feature request related to a problem? Please describe.**
The "keyboardPress method only allows for transmission of an ASCII character. You cannot for example send function keys or as a special keypresses.

**Describe the solution you'd like**
I recommend the current "keyboardPress" method be renamed "keyboardPressASCII" and that "keyboardPress" be implemented as a direct call to "tud_hid_keyboard_report". I realize that will break compatibility with any existing applications. If you don't want to rename keyboardPress to keep compatibility then perhaps the new direct method should be named keyboardPressRaw or something similar.

**Additional context**
Just a general comment I think the code could use a lot more comments or documentation. I would be willing to contribute to that but I'm not that knowledgeable about USB and personally I've only interested in the mouse and keyboard portions of the HID section of the code. I would be willing to collaborate with a more knowledgeable person to add comments and/or write learning guides to further explain how to use this code.

hathach, 2020-04-06:
you can use `keyboardReport()` for a raw HID keycode input. https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/master/src/Adafruit_USBD_HID.h#L55

Note that the keycode is totally different than ascii key, here is the list
https://github.com/hathach/tinyusb/blob/master/src/class/hid/hid.h#L215

keyboardPress() and keyboardRelease() are simply the helper function of the raw keyboardReport(). Initially I didn't want to include these 2, but think maybe it is useful for most users to have a function to covert ascii to keycode.

Indeed the library seriously is lacking documentation, comments and/or tutorial. That is entirely my fault, though I somehow always find myself struggling with other projects :sweat_smile: . Next time I work on this library, I will add more comment for API etc.. Thank you for willingness to contribute, if you have any questions on how the API() or USB work, don't hesitate to ask, I am more than happy to answer as far as I could.

cyborg5, 2020-04-06:
Duh… So sorry I feel like an idiot. I didn't even look at keyboardReport. It's exactly what I was proposing you implement.

But while I've got your attention maybe you can clarify something. Suppose I do a keypress using keyboardReport or keyboardPress and then I press a second key, how do I tell it which one to release or can I only release all pressed keys?

I would also like some clarification about just what is report_id in the entire API. Is it like a handle that my application makes up on its own or does it require some specific values under certain circumstances. Most of the time you leave it zero in the examples. In a combo examples I see you give it different values but I'm not sure exactly what those values specify. Is it an index into the list of multi-functions your particular device has or is that just a made up handle?

closed 2020-04-07 by cyborg5

hathach, 2020-04-09:
though you close the issue, I will answer anyway for people with the same question. `report_id` is part of USB HID specs, it bundles several report (keyboard, mouse, gamepad) into an interface. Therefore report_id is used to indicate which report for which. Report ID is dictated by application in the report descriptor