adafruit/Adafruit_TinyUSB_Arduino #52: Cannot compile   usb_hid.mouseButtonRelease(0);

cyborg5, 2020-04-12:
**Describe the bug**
Adding a statement such as "usb_hid.mouseButtonRelease(0);" causes a compiler error.

C:\Users\Chris.cy\Documents\Arduino\Freedom Wing\my_hid_mouse/my_hid_mouse.ino:72: undefined reference to `Adafruit_USBD_HID::mouseButtonRelease(unsigned char)'

I see in the definition the method is defined as

  static inline bool mouseButtonRelease(uint8_t report_id);

Is there some special syntax when invoking a static inline method?

hathach, 2020-04-14:
ah sorry, there is typo in prototype for that function PR fix is on the way #53 , will merge when ci passes. Thank you for reporting the issue.

hathach, 2020-04-15:
it should be fixed with the current master. Please check to see if that works for you.

commit 950fc6bb (2020-04-15): Merge pull request #53 from adafruit/develop

fix #52 mouseButtonRelease usage
closed 2020-04-15 by hathach