adafruit/Adafruit_TinyUSB_Arduino #59: [Feature] Setting USB PID in user program

PTS93, 2020-04-30:
From what I've seen its not possible to change the PID from within the user program because it is being set/overwritten at compile time through the board definition?

I may also be wrong about that. Still learning how tinyusb is setup.
This would help me a great deal if I could change the PID of a board on the fly without making a new board definition in the core.

todbot, 2020-04-30:
It is possible, via the underlying USBDevice.  Do `USBDevice.setID( VID, PID );`
Here's one example showing it in use
https://github.com/todbot/hidapitester/blob/master/hid_generic_trinketm0/hid_generic_trinketm0.ino

PTS93, 2020-04-30:
Ah thats great! Thanks Tod :)

closed 2020-04-30 by PTS93

hathach, 2020-05-01:
I should do a better job with documentation :smile:

PTS93, 2020-05-01:
@hathach A block diagram of the core code and how it relates to "sub-library" like this one and the Arduino-core part would help a lot already to understand what I need to access or look into for a specific feature :)
For the API something auto generated to search in like doxygen that can be extended with code comments over time sounds good to me! I'm sure the community would help with that once there is a place to start with.

hathach, 2020-05-02:
That is a lots of works, I will try to add more function comment next time I switched back to Arduino TinyUSB. Hopefully it is not a bug hunting work :D

RenaKunisaki, 2023-12-21:
> It is possible, via the underlying USBDevice. Do `USBDevice.setID( VID, PID );` Here's one example showing it in use https://github.com/todbot/hidapitester/blob/master/hid_generic_trinketm0/hid_generic_trinketm0.ino

Link is broken :(

timonsku, 2023-12-21:
> > It is possible, via the underlying USBDevice. Do `USBDevice.setID( VID, PID );` Here's one example showing it in use https://github.com/todbot/hidapitester/blob/master/hid_generic_trinketm0/hid_generic_trinketm0.ino
> 
> Link is broken :(

It's just `TinyUSBDevice.setID( VID, PID );`. That object is available with the default included header.

todbot, 2023-12-21:
What @timonsku said.  But if you're curious, the equivalent file would be: https://github.com/todbot/hidapitester/blob/main/test_hardware/hidtest_tinyusb_static/hidtest_tinyusb_static.ino