adafruit/Adafruit_TinyUSB_Arduino #60: Crashing in version 0.9.0

PTS93, 2020-05-04:
The USB device crashes when uploading any of the examples. I haven't tried all but at least MSC and HID seem to crash.
On SAMD21 tried on Metro M0 and Itsy Bitsy M0

 - Adafruit SAMD Core 1.5.13
 - TinyUSB 0.9.0
 - Windows

hathach, 2020-05-04:
@PTS93 thanks for reporting the issue, I did make a typo with USB IRQhandler on samd, I remembered testing it, but probably before the typo or probably with m4 only. Patch is on the way, we will release it asap.

hathach, 2020-05-04:
The bug is in the core BSP though, it may take a few hours to get online. If you are in hurry, you could just follow the simple modification in this PR to work with https://github.com/adafruit/ArduinoCore-samd/pull/226/files

closed 2020-05-04 by hathach

PTS93, 2020-05-04:
thanks! for now I can work with the prev. version :)