adafruit/Adafruit_TinyUSB_Arduino #62: Library dependencies

nanosonde, 2020-05-07:
Hi!

Is it really necessary to have all those library dependencies listed in the file library.properties?
```
depends=Adafruit SPIFlash, MIDI Library, Adafruit seesaw Library, Adafruit NeoPixel, SdFat - Adafruit Fork, SD, Adafruit Circuit Playground
```
For example PlatformIO tries to pull in all those dependencies although they are not really required for the sketch to compile as it only uses the Adafruit TinyUSB library.

PTS93, 2020-05-07:
not really an option around this as they are needed for the examples, there would need to be an optional dependency field which is not supported atm by Arduino

ladyada, 2020-05-07:
yes - they are required to test the examples. 
there's a way around platformio's dependancy management, @ivankravets can help you! :)

closed 2020-05-07 by ladyada

nanosonde, 2020-05-07:
Ok, then I will have to turn off the LDF automatic mode.