adafruit/Adafruit_TinyUSB_Arduino #63: More supported board, especially Teensy 4.0 [Feature]

isaacjacobson92, 2020-05-12:
This library is a game changer! I'd love to use this library for other projects that utilize other development boards.

Specifically, I'd love to use the SD Card Mass Storage Device feature with a **Teensy 4.0**.

It looks like hathach's original library seems to support Teensy 4.0 (https://github.com/hathach/tinyusb/blob/master/docs/boards.md)

But I can't seem to get any of the example files from Adafruit's TinyUSB Library to work on a Teensy 4.0.

Am I missing something? Is there something I can do to read/write on an SD Card in a Teensy 4.0 via the USB port as a Mass Storage Device like you can with Adafruit's M0 & M4 dev boards?

Is support for Teensy 4.0 with this library something that may be possible in the near future?

Thanks!!!

tannewt, 2020-05-12:
@iosonoisaac What happens when you try to use the examples? Do they compile?

iosonoisaac, 2020-05-12:
When I select the board as a Teensy, I get this error when attempting to compile the "msc_sd" example:

```
.../Documents/Arduino/libraries/Adafruit_TinyUSB_Library/src/Adafruit_USBD_HID.h:28:35: fatal error: Adafruit_TinyUSB_Core.h: No such file or directory
compilation terminated.
Error compiling for board Teensy 4.0.
```

This error is not given when I use one of Adafruit's supported boards.

I'm guessing this is partly because when the board is selected as a Teensy (or anything other than Adafruit's supported boards), I don't have the option to set the "USB Stack" to TinyUSB:
[image]

However, that is an option when I use the ItsyBitsy M4 for example:
[image]

None of the examples in Adafruit's TinyUSB library compile with Teensy 4.0, but I'm specifically trying "msc_sd."

hathach, 2020-05-13:
@iosonoisaac thanks for your kind words, unfortunately TinyUSB requires the core BSP to support (port) it.  It is not as simple as other peripherals such as I2C. Since most the core will use and setup using its USB stack for CDC before setup() is run.  

For BSP core to support TinyUSB  https://github.com/adafruit/Adafruit_TinyUSB_ArduinoCore must be added and ported to the platform. You will need to ask Teensy BSP maintaining team to port it first.

In the future, I will try to see if we could make it easier to support on new platform by moving most parts of the Core to the Lib, more or less it still needs BSP changes to adopt the library.

iosonoisaac, 2020-05-13:
@hathach  thanks for taking the time to respond and give that info. I'll see if anyone from the BSP maintaining team has time to look into it.

hathach, 2020-05-13:
@iosonoisaac here is how samd and nrf implement the core support
- https://github.com/adafruit/ArduinoCore-samd/tree/master/cores/arduino/TinyUSB
- https://github.com/adafruit/Adafruit_nRF52_Arduino/tree/master/cores/nRF5/TinyUSB

Note it is still subject to change though, since it is hard to get the big picture when I added the initial port.

hathach, 2020-08-03:
closed since there is nothing we could do until core team from Teensy support it first.

closed 2020-08-03 by hathach