adafruit/Adafruit_TinyUSB_Arduino #64: MIDI Messages One Message Behind

acumartini, 2020-08-10:
**Describe the bug**
Hey all. Not sure this is an issue with your library or the underlying MIDI Library but want to confirm. I am running the Adafruit TinyUSB MIDI version 0.9.0 and my messages appear to be one step behind.

For example the first message the system sees is a NOTE_ON message, but only after I release the note. So hitting three consecutive keys on a virtual midi keyboard shows up like this in the NoteOn and NoteOff handlers:

15:56:27.083 -> Note on: channel = 1, pitch = 60, velocity - 100
15:56:30.077 -> Note off: channel = 1, pitch = 60, velocity - 0
15:56:30.511 -> Note on: channel = 1, pitch = 62, velocity - 100
15:56:31.779 -> Note off: channel = 1, pitch = 62, velocity - 0
15:56:32.143 -> Note on: channel = 1, pitch = 65, velocity - 100

As you can see I released the last note and the last message is a NOTE_ON type. This output is using your midid_test Example sketch with the MIDI sends commented out. Any ideas or pointers? Should I be bugging the maintainers of MIDI Library?

Note: I am on OSX Catalina using CoreMIDI to connect to the board via USB. Thanks!

**Set up (please complete the following information)**
 - Your Board and BSP version: Feather M4, not sure on BSP
 - TinyUSB library version : 0.9.0
 - Your OS (mac/windows/linux) and its version: Catalina via CoreMIDI
 - Serial debug log if any: see above

**To Reproduce**
Steps to reproduce the behavior:
1.  Run the midi_test example sketch with no MIDI output
2. Connect to your board using OSX CoreMIDI and some kind of Virtual Midi Keyboard. Like "vmpk" (Note: I tested several, all with the same results)
3. Turn on Serial Monitor and start pressing keys
4. See debug output

**Expected behavior**
Pressing a note on the virtual KB should result in a NOTE_ON message when first pressed, then a NOTE_OFF message when released. Multiple notes at one time should behave the same.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

hathach, 2020-08-12:
thanks for bug report, BSP version is required for us to try reproducing the issue, if you are not sure, please use the latest and update the field.

acumartini, 2020-08-12:
Happy top provide the BSP, but sorry I don't actually know what that is or where to find it. Apologeez for noob.

hathach, 2020-08-12:
BSP is the Adruino core SAMD version that you installed using the Boad Manager in the IDE

acumartini, 2020-08-12:
Using the additional URL: https://adafruit.github.io/arduino-board-index/package_adafruit_index.json

Installed BMP: Adafruit SAMD Boards 1.6.0

hathach, 2020-08-20:
thanks for updating the BSP version, I will try to reproduce this issue whenever I could.

acumartini, 2020-08-26:
Update. Tested this on a Windows machine with the all the same board, BSP, etc. and got the same result. So its definitely not related to OS specific MIDI drivers.

deladriere, 2020-08-27:
I have the same problem (on mac)
My USB code works fine is I use 
`#include <USB-MIDI.h>. 
USBMIDI_CREATE_DEFAULT_INSTANCE();`
instead of 
`#include <Adafruit_TinyUSB.h>
#include <MIDI.h>
Adafruit_USBD_MIDI usb_midi;
MIDI_CREATE_INSTANCE(Adafruit_USBD_MIDI, usb_midi, MIDI);
`
`[env:adafruit_feather_m0]
platform = atmelsam@4.5.0
board = adafruit_feather_m0
build_flags = -D USE_TINYUSB
framework = arduino
lib_deps =
    Adafruit TinyUSB Library@0.9.0
    MIDI Library@4.3.1
    SSD1306Ascii@1.2.4
monitor_speed = 115200`

deladriere, 2020-08-28:
@hathach I have question : I use Adafruit_TinyUSB.h just to get  `USBDevice.setProductDescriptor("name");` working
Is there a way to do this and avoid using Adafruit_TinyUSB.h  and all the dependencies?

chee, 2021-01-04:
this is happening to me too, with a Pygamer (which is also an M4).

tried sending the midi signals with VMPK and Bitwig. using linux. 

Adafruit SAMD Boards **1.6.4**
TinyUSB **0.10.1**

if i press a key in VMPK, nothing happens on the board. then when i release the key, the handleNoteOn callback is called. if i press another key, only then is the handleNoteOff callback called.

Kprellz, 2021-04-09:
Similar issue. has this been resolved?

virtuaCode, 2021-08-01:
I have the same problem. I'm using a Raspberry Pi Pico the together with the [Arduino-Pico](https://github.com/earlephilhower/arduino-pico).

hathach, 2021-08-02:
I am finally got time to troubleshoot and fix this issue with PR #114 , please give it a try. This should have been fixed earlier but I have been doing all the rework and merged tinyusb core + lib and get it ported across platform rp2040, and this somehow got off the radar. Glad that I finally got time to do so.

PR #114 (2021-08-03): update tinyusb 
- update to commit https://github.com/hathach/tinyusb/commit/af8e5a90f4244bfe1174724b86bc4b73599b932c 
- fix #64
closed 2021-08-03 by hathach