adafruit/Adafruit_TinyUSB_Arduino #69: HID echo to python example is losing a character (from the buffer?)

SimonMerrett, 2020-11-12:
**Describe the bug**
`hid_generic_inout.ino` example appears to be losing the first character in the buffer. Here is the output from the python script:

```
Openning HID device with VID = 0x239A
{'path': b'\\\\?\\hid#vid_239a&pid_800f&mi_02#8&1f3020af&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}', 'vendor_id': 9114, 'product_id': 32783, 'serial_number': '84C8D28C5150484347202020FF042C23', 'release_number': 256, 'manufacturer_string': 'Adafruit', 'product_string': 'TinyUSB HID Generic', 'usage_page': 65280, 'usage': 1, 'interface_number': 2}
Send text to HID Device : a
Received from HID Device: b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'

Send text to HID Device : a
Received from HID Device: b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'

Send text to HID Device : sa
Received from HID Device: b'a\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'

Send text to HID Device : s
Received from HID Device: b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'

Send text to HID Device : ss
Received from HID Device: b's\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'

Send text to HID Device : sdf
Received from HID Device: b'df\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'

Send text to HID Device : aaaa
Received from HID Device: b'aaa\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'

Send text to HID Device : a
Received from HID Device: b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'

Send text to HID Device : aa
Received from HID Device: b'a\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'

Send text to HID Device : aaa
Received from HID Device: b'aa\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'

Send text to HID Device : Traceback (most recent call last):
  File "hidTest.py", line 15, in <module>
    str_out = input("Send text to HID Device : ").encode('utf-8')
KeyboardInterrupt
```

**Set up (please complete the following information)**
 - ItsyBitsy M0, Adafruit SAMD boards V1.6.4
 - TinyUSB library version V0.10.0
 - Win10 2004

**To Reproduce**
Steps to reproduce the behavior:
1. Run the example and send the ItsyBitsy some text over the python console.

**Expected behavior**
The same text I entered would be written back. Instead, it is missing the first character I type.

SimonMerrett, 2020-11-17:
I might be overthinking it but could the unicode be part of the issue? With the encoding stripping the first character under the presumption that it is the byte order mark (or something to do with unicode encoding/decoding)? https://en.wikipedia.org/wiki/UTF-8#Byte_order_mark

hathach, 2020-11-17:
sorry, I haven't got time to work on this repo just yet. Please be patient and keep updating what you discover. Maybe it is worth to use some low level to capture software to see the actual data transferred. There is a few of them, though I actually don't know one, since It mostly use my usb hardware analyzer for troubleshooting. Thanks for the issue.

SimonMerrett, 2020-11-17:
Yeah don't worry, I can just send a dummy/filler character for now. Will shout if I find anything more useful.

SimonMerrett, 2020-11-17:
I have a couple of contenders from another look around. Most likely it could be something to do with Win 10. Looking at the accompanying .js script with this example [it says](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/a514551c414f29b6d16e5123f72757d528c1f264/examples/HID/hid_generic_inout/hid_test.js#L22) to shift a zero in front of everything sent from Windows. The python doesn't appear to do this. I don't have a chance to send from mac or linux atm and I don't have npm installed on Win10 to my knowledge, so would be great if someone on mac or linux could test for the same issue.

Less likely is the report section of tusb hid but I saw [some buffer size stuff](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/a514551c414f29b6d16e5123f72757d528c1f264/examples/HID/hid_generic_inout/hid_test.js#L22) that might be worth double checking (ie report id 0) in `tud_hid_n_report`.

keyz182, 2021-02-14:
Got a minimal reproduction here - https://github.com/keyz182/tinyusb-report-first-byte if someone wants to look. I'm testing on an Adafruit QT Py if that matters. 

Compile and upload with PlatformIO, then run the python `hidtest.py` script, and enter some text. Always cuts off the first byte as above.

Based on [this](https://github.com/node-hid/node-hid/issues/187) however, it looks like it may be a bug in the py library / quirk of the Windows USB stack. I'm no expert however, so will leave that decision to those more knowledgeable. For now, I'll use the work-around to prepend a zero if on windows.

todbot, 2021-02-14:
The first byte of the buffer is where the HID reportId lives.  If your device does not use reportIds in its HID Report Descriptor, then this first byte should be 0.

SimonMerrett, 2021-02-14:
> If your device does not use reportIds in its HID Report Descriptor, then this first byte should be 0.

Thanks but does my device, an Adafruit itsybitsy M0 running the provided example use them or not? The meagre explanation / documentation is leaving me quite confused about whether this is supposed to happen or I'm doing something wrong. Examples should probably explain nuances if they don't work perfectly in every platform/scenario.

todbot, 2021-02-14:
For the `hid_generic.ino`example, it is not using reportIds, so that byte should be zero.

SimonMerrett, 2021-02-14:
Thanks, I'm on mobile not pc but it would be good to have that commented in at least the python code if not the .ino.

Is it the OS usb interface or tusb which is stripping the first character?

hathach, 2021-08-06:
Sorry for the late response, I finally tested this out and confirm this issue on windows, (no issue with Linux though). For a bit of hid info, 
- when report ID is used (dictated by report descriptor): reportID is first byte of the report payload. However
- when  there is no reportID, in case of hid_generic_inout: report id is dropped entirely (not zero). Therefore 1st byte is an actual payload. 

It looks to me that linux backend handle it well but windows 10 backend probably mixed it up. I will try to see if we could fix it changing the script. Adding reportID by firmware is also an option, but not ideal, since it reduce payload to 63, whereas 64 is kind of the right number of payload for a HID message.

Anyway, thank you for your patient.

hathach, 2021-08-09:
@todbot is spotted on, HIDAPI to normalize the single and composite report, required that function call must preceeded with report ID where single report use dummy 0x00 as report ID (which isn't sent to usb device). I will update the hid_test.py script and make PR to fix this.

PR #120 (2021-08-09): Fix hid inout script example
fix #69 , HIDAPI requires reportID always preceded even with single report device (dummy reportID = 0x00) more info : https://github.com/libusb/hidapi/blob/master/hidapi/hidapi.h#L230
closed 2021-08-09 by hathach