adafruit/Adafruit_TinyUSB_Arduino #70: WebUSB demo html page 404

jpliew, 2020-12-26:
Link referred from https://learn.adafruit.com/using-webusb-with-arduino-and-tinyusb/serial-communications-example

going to https://adafruit.github.io/Adafruit_TinyUSB_Arduino/examples/webusb-serial

producing a 404

hathach, 2020-12-27:
thank for reporting the issue, this is fixed. Somehow the repo pages setting is changed from `master/root` to `gh-pages` (which isn't hosting the example).

closed 2020-12-27 by hathach

jpliew, 2020-12-28:
still 404

hathach, 2020-12-28:
please add trailing `/` to the url,  github page probably change how they handle url, ` https://adafruit.github.io/Adafruit_TinyUSB_Arduino/examples/webusb-serial/` , or better have explicitly add index.html to the url as well. I have updated the example with index.html and will update the learn guide as well.