adafruit/Adafruit_TinyUSB_Arduino #74: MIDI Echo

Kprellz, 2021-03-26:
Hey ya'll, 
Seem to be experiencing an issue where any incoming Midi message gets echo'd back to the sender. Is there a way to prevent this? or is this a bug of some kind?
ItsyBitsy M0
mac osx 10.15.5
even the example program does this. 

attached is a screen shot of MIDI Monitor reading midi messages to and from the device. I wrote a simple code to just send out a note on ever second or so.  the only code in my Loop is MIDI.read();

[image]

Kprellz, 2021-03-26:
Also tested on a Metro M0 Express to the same results.

Kprellz, 2021-04-06:
Figured this one out actually. anyone having a similar issue can simply due MIDI.turnThruOff();

closed 2021-04-06 by Kprellz

dkadrios, 2024-08-25:
@Kprellz 
cheers mate, you saved me from pulling out more hair :)