adafruit/Adafruit_TinyUSB_Arduino #75: [Feature] RP2040 Support

Github73840134, 2021-04-05:
Hello there,
I have seen this library and the other tinyUSB library, The non adafruit one is compatible with the raspberry pi pico RP2040. Although the documentation does not list it. I am wondering due to the similarities of the two libraies would this one be compatible with the RP2040? I prefer this one due to its simplicity.

Thanks

todbot, 2021-04-05:
This repo is an Arduino library that wraps the TinyUSB C library, making it usable for those Arduino-compatible boards that use TinyUSB in their Arduino board package. I believe currently this is just a few Adafruit boards.

ladyada, 2021-04-05:
there's no official arduino support yet, so this library could not be used. hopefully arduino support will exist, and they may use this code!

closed 2021-04-05 by ladyada