adafruit/Adafruit_TinyUSB_Arduino #76: Cannot compile for Adafruit Feather M4 CAN (SAME51)

Callum6677, 2021-04-06:
Any time I compile with tinyUSB included, I simply get a nice failure to build with the following setup:
 - Adafruit Feather M4 CAN (SAME51)
 - Adafruit TinyUSB Library 0.10.1 
 - Arduino IDE 1.8.13
 - Windows 10 2002 19041
 - Any of the included TinyUSB samples
 - Board set to use TinyUSB stack

`
Arduino: 1.8.13 (Windows 10), Board: "Adafruit Feather M4 CAN (SAME51), Enabled, 120 MHz (standard), Small (-Os) (standard), 50 MHz (standard), TinyUSB, Off"

C:\Users\callu\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.6.5\cores\arduino\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\portable\microchip\samd\dcd_samd.c:39:27: error: unknown type name 'UsbDeviceDescBank'

   39 | static TU_ATTR_ALIGNED(4) UsbDeviceDescBank sram_registers[8][2];

      |                           ^~~~~~~~~~~~~~~~~

C:\Users\callu\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.6.5\cores\arduino\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\portable\microchip\samd\dcd_samd.c: In function 'prepare_setup':

C:\Users\callu\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.6.5\cores\arduino\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\portable\microchip\samd\dcd_samd.c:46:23: error: request for member 'ADDR' in something not a structure or union

   46 |   sram_registers[0][0].ADDR.reg = (uint32_t) _setup_packet;

      |                       ^

C:\Users\callu\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.6.5\cores\arduino\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\portable\microchip\samd\dcd_samd.c:47:23: error: request for member 'PCKSIZE' in something not a structure or union

   47 |   sram_registers[0][0].PCKSIZE.bit.MULTI_PACKET_SIZE = sizeof(_setup_packet);

      |                       ^

C:\Users\callu\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.6.5\cores\arduino\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\portable\microchip\samd\dcd_samd.c:48:23: error: request for member 'PCKSIZE' in something not a structure or union

   48 |   sram_registers[0][0].PCKSIZE.bit.BYTE_COUNT = 0;

      |                       ^

C:\Users\callu\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.6.5\cores\arduino\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\portable\microchip\samd\dcd_samd.c: In function 'bus_reset':

C:\Users\callu\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.6.5\cores\arduino\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\portable\microchip\samd\dcd_samd.c:55:3: error: unknown type name 'UsbDeviceDescBank'

   55 |   UsbDeviceDescBank* bank_out = &sram_registers[0][TUSB_DIR_OUT];

      |   ^~~~~~~~~~~~~~~~~

C:\Users\callu\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.6.5\cores\arduino\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\portable\microchip\samd\dcd_samd.c:56:11: error: request for member 'PCKSIZE' in something not a structure or union

   56 |   bank_out->PCKSIZE.bit.SIZE = 0x3;

      |           ^~

C:\Users\callu\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.6.5\cores\arduino\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\portable\microchip\samd\dcd_samd.c:57:3: error: unknown type name 'UsbDeviceDescBank'

   57 |   UsbDeviceDescBank* bank_in = &sram_registers[0][TUSB_DIR_IN];

      |   ^~~~~~~~~~~~~~~~~

C:\Users\callu\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.6.5\cores\arduino\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\portable\microchip\samd\dcd_samd.c:58:10: error: request for member 'PCKSIZE' in something not a structure or union

   58 |   bank_in->PCKSIZE.bit.SIZE = 0x3;

      |          ^~

C:\Users\callu\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.6.5\cores\arduino\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\portable\microchip\samd\dcd_samd.c:60:3: error: unknown type name 'UsbDeviceEndpoint'

   60 |   UsbDeviceEndpoint* ep = &USB->DEVICE.DeviceEndpoint[0];

      |   ^~~~~~~~~~~~~~~~~

C:\Users\callu\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.6.5\cores\arduino\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\portable\microchip\samd\dcd_samd.c:60:28: error: 'USB' undeclared (first use in this function)
[... 593 lines cut ...]
  373 |   if ( int_status & USB_DEVICE_INTFLAG_EORST )

      |                     ^~~~~~~~~~~~~~~~~~~~~~~~

C:\Users\callu\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.6.5\cores\arduino\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\portable\microchip\samd\dcd_samd.c:393:51: error: 'USB_DEVICE_EPINTFLAG_RXSTP' undeclared (first use in this function)

  393 |     USB->DEVICE.DeviceEndpoint[0].EPINTFLAG.reg = USB_DEVICE_EPINTFLAG_RXSTP | USB_DEVICE_EPINTFLAG_TRCPT0;

      |                                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~

C:\Users\callu\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.6.5\cores\arduino\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\portable\microchip\samd\dcd_samd.c:393:80: error: 'USB_DEVICE_EPINTFLAG_TRCPT0' undeclared (first use in this function)

  393 |     USB->DEVICE.DeviceEndpoint[0].EPINTFLAG.reg = USB_DEVICE_EPINTFLAG_RXSTP | USB_DEVICE_EPINTFLAG_TRCPT0;

      |                                                                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~

exit status 1

Error compiling for board Adafruit Feather M4 CAN (SAME51).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

`

Callum6677, 2021-04-06:
Apparently I cannot use github styling properly. Apologies.
[buildOutput.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/6265387/buildOutput.txt)

hathach, 2021-04-06:
> Apparently I cannot use github styling properly. Apologies.
> [buildOutput.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/6265387/buildOutput.txt)

it is tripple backslash, though the log is too long anyway, it is best to be an txt. I will try to check this out asap.

hathach, 2021-04-07:
I am able to reproduce the issue, it is probably due to the slight difference in macro SAMD51 SAME51 and/or other include config.  Will make an pr to fix this soon enough

hathach, 2021-04-08:
@ladyada this issue is more involved than initial thought. The root issue is tinyusb naturally include "sam.h" which has defined of `__SAME51J19A__` which will include the correct mcu header. The issue is SAME51 is not supported by the CMSIS-Atmel package https://github.com/adafruit/ArduinoModule-CMSIS-Atmel/blob/master/CMSIS-Atmel/CMSIS/Device/ATMEL/sam.h

We tried to put the chip definition into the variant folder, however, since the tool cmsis path has higher priority in the include order (since it appears first) it is always got include. We can re-order the the include path. However, I think the proper fix should be moving mcu header into the cmsis-atmel. I will make an PR to official arduino repo, meanwhile we can use the our own tool release (possibly 1.2.1 for the next release). This will also make it much easier to add more board with same51.

hathach, 2021-04-08:
PR to add SAME51 header to official package is on the way https://github.com/arduino/ArduinoModule-CMSIS-Atmel/pull/16

closed 2021-04-10 by hathach

hathach, 2021-04-11:
should be fixed by now, can you install the new bsp 1.6.7 and try again.

Callum6677, 2021-04-12:
Working great now, that was a quick turn around, cheers