adafruit/Adafruit_TinyUSB_Arduino #78: Can't use Keyboard Modifier + Mouse Scroll Wheel at the same time

Jorgen-VikingGod, 2021-04-18:
**Describe the bug**
I want to send CTRL+ScrollWheel or SHIFT+ScrollWheel to control some programs like KiCAD etc.
But I can't find any description or example of how to send Keyboard Modifier and Mouse Scroll Wheel combination...

**Set up (please complete the following information)**
 - TinyUSB library (github latest) 
 - Your OS (linux) and its version: Pop!_OS 20.10

**Expected behavior**
Simple example of how to use Keyboard and Mouse combination

**Additional context**
I need this to finish my software migration from old keyboard library to TinyUSB for my custom MacroPad

Thx

hathach, 2021-04-19:
this is not a bug, for question please post on forum support.

closed 2021-04-19 by hathach

Jorgen-VikingGod, 2021-04-19:
I already post in the forum too - but no response - so I wonder if it is possible with TinyUSB or not?

hathach, 2021-04-19:
can you post link to the forum, I will try to answer there.

Jorgen-VikingGod, 2021-04-19:
sure - thx @hathach  https://forums.adafruit.com/viewtopic.php?f=25&t=177940&p=866517