adafruit/Adafruit_TinyUSB_Arduino #81: On Raspberry Pico in Platformio cannot compile 'cause Adafruit_TinyUSB_Core.h is missed but mentioned

katurov, 2021-05-10:
**Set up**

- MacOS X 10.15.7, PlatformIO (core 5.1.1)
- Raspberry Pico
- adafruit/Adafruit TinyUSB Library@^0.10.5

**Description**

Tried to build project with clear example (echo), got problem:

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (1.0.0) > Raspberry Pi Pico
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, raspberrypi-swd)
PACKAGES: 
 - framework-arduino-mbed 2.0.0 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 64 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Adafruit TinyUSB Library> 0.10.5
Building in release mode
Compiling .pio/build/pico/src/main.cpp.o
Compiling .pio/build/pico/lib88e/Adafruit TinyUSB Library/Adafruit_TinyUSB.cpp.o
Compiling .pio/build/pico/lib88e/Adafruit TinyUSB Library/Adafruit_USBD_HID.cpp.o
Compiling .pio/build/pico/lib88e/Adafruit TinyUSB Library/Adafruit_USBD_MIDI.cpp.o
In file included from .pio/libdeps/pico/Adafruit TinyUSB Library/src/Adafruit_USBD_HID.cpp:25:
.pio/libdeps/pico/Adafruit TinyUSB Library/src/Adafruit_USBD_HID.h:28:10: fatal error: Adafruit_TinyUSB_Core.h: No such file or directory

*******************************************************************************
* Looking for Adafruit_TinyUSB_Core.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Adafruit_TinyUSB_Core.h"
* Web  > https://platformio.org/lib/search?query=header:Adafruit_TinyUSB_Core.h
*
*******************************************************************************

**To Reproduce**

Steps to reproduce the behavior:
1. Create new project for Pico
2. Add library Adafruit TinyUSB Library via standard interface from Home->Libraries
3. Use stdio_init_all(); in setup() section
4. Try to compile

ladyada, 2021-05-10:
the arduino core needs to support tinyusb in the Core but right now it doesnt, so its not possible to fix this. you can ask arduino to please help get the support going that would be good!

closed 2021-05-11 by katurov

hathach, 2021-05-11:
the `Adafruit_TinyUSB_Core` will be obsolete soon once #80 is merged, you could get more info by reading that PR and checking out its branch `major-v1` README