adafruit/Adafruit_TinyUSB_Arduino #83: Add option to remove CDC ACM interface

gdsports, 2021-05-18:
I propose adding a compile time option to remove the CDC ACM port. This allows
Adafruit boards to more closely emulate real USB keyboards, joysticks, etc. The
IAC class descriptor and CDC ACM descriptors confuse some embedded devices such
as KVM switches, old computer BIOS, game consoles, etc.

The PluggableUSB stack has this option. It requires some code patching but
it does work. The symbol is CDC_ENABLED.

I propose using CDC_DISABLED so when it is not defined (the default), the
behavior is unchanged. The symbol can be defined by changing the boards.txt

`adafruit_qtpy_m0.menu.usbstack.tinyusb.build.flags.usbstack=-DUSE_TINYUSB -DCDC_DISABLED`

I use arduino-cli so it is possible to define the symbol without changing
boards.txt. The VID, PID, and other USB parameters can be passed via command
line options so exact emulation is much easier.

The following change applies to this repo.

```
diff --git a/Adafruit_USBD_Device.cpp b/Adafruit_USBD_Device.cpp
index 1f51925..458c726 100644
--- a/Adafruit_USBD_Device.cpp
+++ b/Adafruit_USBD_Device.cpp
@@ -60,11 +60,17 @@ Adafruit_USBD_Device::Adafruit_USBD_Device(void)
     .bDescriptorType    = TUSB_DESC_DEVICE,
     .bcdUSB             = 0x0200,
 
+#ifdef CDC_DISABLED
+    .bDeviceClass       = 0,
+    .bDeviceSubClass    = 0,
+    .bDeviceProtocol    = 0,
+#else
     // Use Interface Association Descriptor (IAD) for CDC
     // As required by USB Specs IAD's subclass must be common class (2) and protocol must be IAD (1)
     .bDeviceClass       = TUSB_CLASS_MISC,
     .bDeviceSubClass    = MISC_SUBCLASS_COMMON,
     .bDeviceProtocol    = MISC_PROTOCOL_IAD,
+#endif
 
     .bMaxPacketSize0    = CFG_TUD_ENDPOINT0_SIZE,
```

Changes are required to the SAMD and nrf52 cores to conditionalize the call to
addInterface based on !CDC_DISABLED.

https://github.com/adafruit/ArduinoCore-samd/blob/master/cores/arduino/TinyUSB/Adafruit_TinyUSB_SAMD.cpp#L81

https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/cores/nRF5/TinyUSB/Adafruit_TinyUSB_nRF.cpp#L94

I do not have an nRF52840 board around at the moment but have tested the changes
out on an ItsyBitsy M0.

guruthree, 2021-07-17:
Is there any documentation on disabling the CDC serial device? The links above in https://github.com/adafruit/Adafruit_TinyUSB_Arduino/issues/83#issuecomment-834149951 are now 404 not found. The only way I've been able to figure out disabling the serial device is to create a dummy header file with an `Adafruit_USBD_CDC` class that only contains an empty begin method and a `#define ADAFRUIT_USBD_CDC_H_` to keep the real file from being included later.

todbot, 2021-07-18:
@guruthree that example is in the main `examples` directory: https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/master/examples/CDC/no_serial/no_serial.ino
I believe you'll also find in the Arduino IDE under File->Examples->Adafruit_TinyUSB_Arduino if you've installed this library

oomek, 2022-05-20:
`Serial.end()` did not work for me sadly.

[image]

How do I get rid of these 2 devices?

hathach, 2022-05-20:
Please open your own issue with detailed info. This is closed long time ago.