adafruit/Adafruit_TinyUSB_Arduino #86: Issue with devices behind a USB hub

kaysievers, 2021-05-18:
With commit https://github.com/adafruit/Adafruit_TinyUSB_ArduinoCore/commit/8096c8a8f8aad8458ff5762c990e10ae9dad1a1b the devices seem no longer working when they are connected behind a USB hub. Directly connecting them to the host's root hub works fine.

Reverting the commit makes the devices appear.

This is a kernel log from Linux (Windows shows similar problems):
```
[34324.981477] usb 1-3: new high-speed USB device number 120 using xhci_hcd
[34325.122916] usb 1-3: New USB device found, idVendor=0bda, idProduct=5411, bcdDevice= 1.21
[34325.122921] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[34325.122924] usb 1-3: Product: 4-Port USB 2.0 Hub
[34325.122927] usb 1-3: Manufacturer: Generic
[34325.126468] hub 1-3:1.0: USB hub found
[34325.128779] hub 1-3:1.0: 4 ports detected
[34325.416491] usb 1-3.1: new full-speed USB device number 121 using xhci_hcd
[34325.492536] usb 1-3.1: device descriptor read/64, error -71
[34325.671522] usb 1-3.1: device descriptor read/64, error -71
[34325.855503] usb 1-3.1: new full-speed USB device number 122 using xhci_hcd
[34325.931543] usb 1-3.1: device descriptor read/64, error -71
[34326.111541] usb 1-3.1: device descriptor read/64, error -71
[34326.219655] usb 1-3-port1: attempt power cycle
[34326.815518] usb 1-3.1: new full-speed USB device number 123 using xhci_hcd
[34326.829286] usb 1-3.1: device descriptor read/8, error -32
[34326.945371] usb 1-3.1: device descriptor read/8, error -32
[34327.128514] usb 1-3.1: new full-speed USB device number 124 using xhci_hcd
[34327.143060] usb 1-3.1: device descriptor read/8, error -32
[34327.265333] usb 1-3.1: device descriptor read/8, error -32
[34327.371687] usb 1-3-port1: unable to enumerate USB device
```

kaysievers, 2021-05-22:
Here is a minimal Arduino sketch which produces the failure. Blocking in setup() until USB is ready is not an option, the device needs to work without USB connected too. And it would also fail the same way if the host triggers a reset/reenumeration of the connected device.

```c
#include <Adafruit_TinyUSB.h>

Adafruit_USBD_MIDI usb;

void setup() {
  usb.begin();

  // Add this to make it work
  //delay(500);
}

void loop() {
  uint8_t packet[4];
  usb.readPacket(packet);
}
```

This is the host's kernel log receiving corrupted messages from the device:
```
[  250.428136] usb 1-6: new full-speed USB device number 38 using xhci_hcd
[  250.440293] usb 1-6: device descriptor read/8, error -71
[  250.558831] usb 1-6: device descriptor read/8, error -71
[  250.665745] usb usb1-port6: unable to enumerate USB device
```

kaysievers, 2021-05-22:
Haha, wrong bug, sorry. But it still applies here too :)

hathach, 2021-05-22:
> Haha, wrong bug, sorry. But it still applies here too :)

No problems, we would need to test this issue again when issue in the main stack repo is fixed

PR #98 (2021-06-20): Update tinyusb to latest with various bug fix
Update tinyusb to latest with various bug fix across ports
- fix #86 when calling midi API when device is not fully enumerated
- fix Serial connection issue with nrf52 on windows
- Update device driver of rp2040 to match tinyusb upstream
- Added feather rp2040 to ci build (skipped the external flash example for now due to SdFat compilation error)
- Add optional debug log for stack with Serial1 on samd core (will expand later to other core).
- lost more bug fixes to the stack
closed 2021-06-20 by hathach