adafruit/Adafruit_TinyUSB_Arduino #90: Please provide all details at least for Setup/Describe/Reproduce

ufanders, 2021-05-23:
**Set up**

- Your OS (mac/windows/linux) and Arduino IDE version
- OSX Mojave 10.14.6, Arduino 1.8.13
- Your Board and BSP version
- Adafruit nRF52840 Feather Express, 0.22.0
- TinyUSB library version
1.0.0

**Describe**

The device does not appear in Finder as a MSC. The code seems to compile fine, and I see expected JEDEC ID and capacity message in the console. It also does not appear in Disk Utility.

I copied in the relevant bits of msc_external_flash.ino. Declaration:
```
//Begin USB Mass Storage (MSC)
#include "SPI.h"
#include "SdFat.h"
#include "Adafruit_SPIFlash.h"
#include "Adafruit_TinyUSB.h"

Adafruit_FlashTransport_QSPI flashTransport;
Adafruit_SPIFlash flash(&flashTransport);

// file system object from SdFat
FatFileSystem fatfs;
FatFile root;
FatFile file;

// USB Mass Storage object
Adafruit_USBD_MSC usb_msc;

// Set to true when PC write to flash
bool changed;
//End USB Mass Storage (MSC)
```

Setup:

```
 //USB MSC and QSPI flash
  if (!flash.begin()) {
    Serial.println("Error, failed to initialize flash chip!");
    while(1) yield();
  }
  Serial.print("Flash chip JEDEC ID: 0x"); Serial.println(flash.getJEDECID(), HEX);
  Serial.print("Flash size: "); Serial.println(flash.size());
  // Set disk vendor id, product id and revision with string up to 8, 16, 4 characters respectively
  usb_msc.setID("Anders", "Pendant", "0.1");
  usb_msc.setReadWriteCallback(msc_read_cb, msc_write_cb, msc_flush_cb);
  usb_msc.setCapacity(flash.size()/512, 512);
  usb_msc.setUnitReady(true);
  usb_msc.begin();

  fatfs.begin(&flash);
```

I don't have any statements in loop() that operate on the flash device, filesystem, or any GPIO thereof. I do however have the USB Serial connection (through no work on my part) as I can interact with the Serial console.

Any help is greatly appreciated!

**To Reproduce**

Steps to reproduce the behavior:
1. Connect USB port to computer or reset the device while connected.

**Screenshots**

DIsk Utility SS here:
[image]

**Log**

15:23:16.647 -> Flash chip JEDEC ID: 0xC84015
15:23:16.647 -> Flash size: 2097152

ufanders, 2021-05-23:
Hmmm, it seems my Arduino Core has version 0.9.0 of this library. I renamed the existing "Adafruit_TinyUSB_Arduino" folder, and copied in the contents of "Adafruit_TinyUSB_Library" v1.0.0. It builds fine, but the effect is the same.

Adding "Serial.end()" also doesn't seem to affect behavior - Serial.print() statements still seem to execute when called.

Shrug!

ufanders, 2021-05-23:
FWIW, compiler says:

```
Multiple libraries were found for "Adafruit_TinyUSB.h"
 Used: /Users/ufanders/Documents/Arduino/libraries/Adafruit_TinyUSB_Library
 Not used: /Users/ufanders/Library/Arduino15/packages/adafruit/hardware/nrf52/0.22.0/libraries/Adafruit_TinyUSB_Arduino
 Not used: /Users/ufanders/Library/Arduino15/packages/adafruit/hardware/nrf52/0.22.0/libraries/Adafruit_TinyUSB_Arduino.old
Using library Bluefruit52Lib at version 0.21.0 in folder: /Users/ufanders/Library/Arduino15/packages/adafruit/hardware/nrf52/0.22.0/libraries/Bluefruit52Lib 
Using library Adafruit_TinyUSB_Library at version 1.0.1 in folder: /Users/ufanders/Documents/Arduino/libraries/Adafruit_TinyUSB_Library 
```

hathach, 2021-05-24:
it is possible that your external flash on nrf52840 isn't formatted correctly yet. Please try
1. run SdFat_format.ino to format it first https://github.com/adafruit/Adafruit_SPIFlash/tree/master/examples/SdFat_format
2. then re run the example to see if it would show up.

ufanders, 2021-05-24:
Hi, thanks for your response.

Does a MSC require a filesystem? I thought it was a block-type controller, operating below any filesystem. I e. the host OS would handle filesystem operations?

hathach, 2021-05-24:
You can format it with disk utilities from macos, but it need to be formatted before show up in the file explorer. Also are you running the example as it is without modification  https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/master/examples/MassStorage/msc_external_flash/msc_external_flash.ino

brentru, 2021-06-30:
You may want to attempt the MSC_External_Flash.ino sketch with the following line before `flash.begin()`:

```
  USBDevice.detach();
  delay(500); // wait for detach
```

and the following lines after `usb_msc.begin()`:
```
  // re-attach the usb device
  USBDevice.attach();
  // wait for enumeration
  delay(500);
```

This causes the filesystem to mount reliably in macOS Finder. @hathach  Should I submit a PR for this?

hathach, 2021-07-01:
thanks @brentru, though this may be another issue. For re-enumeration, I will do a more systematic way later on to demonstrate why and when user should use that. It seems to be a common problem when using initialize lots of other stuffs.

hathach, 2021-07-22:
closed due to lack of response from user

closed 2021-07-22 by hathach