adafruit/Adafruit_TinyUSB_Arduino #93: remote wakeup does not work with nRF52840

ogatatsu, 2021-05-29:
**Set up**

- Your OS (mac/windows/linux) and Arduino IDE version
Windows 10 and Arduino 1.8.15
- Your Board and BSP version
Adafruit nRF52840 Feather Express, 0.22.1
- TinyUSB library version
1.0.3

**Describe**
remote wakeup does not work on feather nRF52840, it works on Itsy Bitsy M0. 

**To Reproduce**

Steps to reproduce the behavior:
1. Write a sketch of hid_mouse.ino on the board.
2. Press UserSw and make sure the pointer moves.
3. Put the PC to sleep.
4. Press UserSw.

hathach, 2021-05-31:
ah right, I don't remember tested this before on nrf. Will check it later on.

hathach, 2021-08-16:
@ogatatsu https://github.com/adafruit/Adafruit_TinyUSB_Arduino/pull/123 fully implement suspend, resume and remote wakeup for nrf5x. Would you mind trying it out !!

PR #123 (2021-08-17): add nrf5x suspend, resume, remote wakeup support
- update tinyusb to https://github.com/hathach/tinyusb/commit/02277133aea7a5d7c0908531c81fa3c3978bc318 which has suspend resume & remote wakeup support
- fix #93
closed 2021-08-17 by hathach