adafruit/Adafruit_TinyUSB_Arduino #94: Sending HID report one after another causes second report to be dropped

timonsku, 2021-06-01:
**Set up**

- Windows 10 20H2 Arduino 1.8.13
- Adafruit Itsy Bitsy M0
- TinyUSB 1.0.3

**Describe**

When sending two ore more usb_hid.sendReport() after another only the first call will actually be sent, all subsequent messages will be dropped and the call returns 0.
Adding a delay of 1ms between each call fixes that which suggests that it was trying to send while the port was still busy but also for some reason didn't trigger this assert in `usbd_edpt_xfer()` at `usbd.c`
`TU_ASSERT(_usbd_dev.ep_status[epnum][dir].busy == 0);`
1ms is exactly the polling rate I have set (`setPollInterval(0)` so in case of Full Speed 1ms).
I'm calling this within the loop(), I would assume that TinyUSB is either waiting for the port to not be busy any more before sending and then returning or throwing an error to tell me to handle it on the user code side.

**To Reproduce**

Execute usb_hid.sendReport() twice in succession within `loop()`, the second call will return 0.
Add delay of 1ms between the two calls and the second call will return 1.

hathach, 2021-06-01:
This is intended behavior unlike cdc, hid is not buffered since it will only be polled by host 1 packet per n ms. Application if needed should buffer the data itself. You could either actively check the tud_hid_ready(), or use the callback tud_hid_report_complete_cb()

https://github.com/hathach/tinyusb/blob/master/examples/device/hid_composite/src/main.c#L226

hathach, 2021-06-01:
For the proposal of blocking, it is never efficient  to block API, the typical polling rate for hid is normally 10 ms which is quite a lot. This type of decision should be done at application level.

timonsku, 2021-06-01:
That is fair, blocking by default would be bad. Checking for tud_hid_ready() would be also be totally fine. I think it then maybe just needs to be communicated. I also noticed I need to call yield() when checking for the ready state in a while loop, otherwise the device just hangs forever.
I think that is all in all fine that the user need to do that, I guess its just something that needs to be documented at some point, otherwise its not very clear what a 0 being returned exactly means :)

closed 2021-06-01 by timonsku

hathach, 2021-06-01:
Yeah, better document is always what I want. But in reality, I don't find time to do it. I has a huge list of pending issue across projects. Please check the code for now. 

yield() and delay() will call tud_task() in the background so yeah when you need any event processing from tinyusb. You need to call that.

timonsku, 2021-06-01:
I feel you :)
I hope I can find the time eventually to dig deeper into the tinyusb code base and help out with some of the tasks

hathach, 2021-06-02:
I am looking forward to your PRs :)