adafruit/Adafruit_TinyUSB_Arduino #95: TinyUSB Debugging doesn't work in the Arduino library

WestfW, 2021-06-09:
In the Adafruit SAMD core ( https://github.com/adafruit/ArduinoCore-samd/blob/master/cores/arduino/tusb_config.h ), there is an option to turn on TinyUSB debugging:'

```
#define CFG_TUSB_DEBUG              0
#if CFG_TUSB_DEBUG
  #define tu_printf                 serial1_printf
  extern int serial1_printf(const char *__restrict __format, ...);
#endif
```

however, if I try to turn this on, I get assorted warnings that tusb_common.h is redefining the macro back to a plain printf() here: https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/master/src/common/tusb_common.h#L300

I can get around that by defining CFG_TUSB_DEBUG_PRINTF as well, but then the errors get more serious, since "Serial1.printf()" is a C++ function, and can't be called directly from the tusb code (which is all just plain C.)

hathach, 2021-06-09:
yeah, it is an leftover from last time I troubleshooting the issue with the lib. Though I have an plan to have a clean debugging later on.

hathach, 2021-06-09:
hmm the serial1_printf code still in the code base https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/master/src/arduino/ports/Adafruit_TinyUSB_samd.cpp#L62
 
You should be able to add this line to the `tusb_config.h` to enable debugging. The serial1_printf should be able to find Serial1 since it is defined in cpp file (with extern C).

```C
#define CFG_TUSB_DEBUG_PRINTF serial1_printf
````

WestfW, 2021-06-10:
Workaround.  Add code to redirect stdout to Serial1 in the setup() in your sketch (in the .ino/C++ file!)
(the following seems to be the (a?) correct incantation to redirect stdout given whatever the current samd cores use for libc...)

```
#include <stdio.h>

int myWrite(void *cookie, const char *buf, int n)    {  // wrapper function
  Uart *s = (Uart *)cookie;
  return s->write((uint8_t*)buf, n);
}

void setup() {
  Serial1.begin(115200);
  stdout = funopen((void *)(&Serial1), NULL, myWrite, NULL, NULL);
  setlinebuf(stdout);
  printf("\n\rSerial1 initialized as stdout for printf\r\n");
  // additional setup code...
```

hathach, 2021-06-10:
Sure there is more way to implement this, the code I used it for troubleshooting previously is above. In future, I will try to make it more core-independent.

Closed this issue since it is not a bug

closed 2021-06-10 by hathach