adafruit/Adafruit_TinyUSB_Arduino #96: Please provide all details at least for Setup/Describe/Reproduce

WestfW, 2021-06-10:
**Set up**
Arduino IDE 1.8.13 (Windows 10), 1.8.15 (MacOS 10.13.6)
Adafruit SAMD boards 1.7.2 ("latest" as of this date.)
Adafruit TinyUSB Library 1.0.3
Example program hid_generic_inout_ramdisk.ino

**Describe**
**To Reproduce**
Is there some sort of timing issue in TinyUSB, perhaps "must start within 0.n seconds of reset/initialization)?

If I insert a `delay(1000);` call as the first thing in setup() (and no other changes), the HID and MSC USB devices do no show up (I do still get the CDC "Serial" port.)

hathach, 2021-06-10:
can you post your setup() code

WestfW, 2021-06-10:
It is literally exactly the code from the example:

```
// the setup function runs once when you press reset or power the board
void setup()
{
#ifdef DEFINE_TO_CAUSE_FAILURE
  delay(1000);
#endif
  usb_msc.setID("Adafruit", "Mass Storage", "1.0");
  
  // Set disk size
  usb_msc.setCapacity(DISK_BLOCK_NUM, DISK_BLOCK_SIZE);

  // Set callback
  usb_msc.setReadWriteCallback(msc_read_cb, msc_write_cb, msc_flush_cb);

  // Set Lun ready (RAM disk is always ready)
  usb_msc.setUnitReady(true);
  
  usb_msc.begin();
  
  usb_hid.enableOutEndpoint(true);
  usb_hid.setPollInterval(2);
  usb_hid.setReportDescriptor(desc_hid_report, sizeof(desc_hid_report));
  usb_hid.setReportCallback(get_report_callback, set_report_callback);

  usb_hid.begin();

  Serial.begin(115200);
  Serial.println("Waiting for USBDevice mount");
  // wait until device mounted
  while( !USBDevice.mounted() ) delay(1);

  Serial.println("Adafruit TinyUSB HID Generic In Out example");
}
```

hathach, 2021-06-10:
Class driver is populated dynamically when it calls`.begin()`. When setup() is called, the enumeration isn’t completely yet. You have to quickly called other class begin such as mac, hid if used. So that this driver is registered and returned to host. Once device enumerated calling those class begin wont change it. You need to detach and attach to force host to re-enumerate.

The enum time is normally around 500-1000 ms for reference

closed 2021-06-10 by hathach

WestfW, 2021-06-10:
Cool.  So I can do the disconnect/detach early in setup to prevent the first try, and then restart things when I'm done with my potentially lengthy "other initialization",  something like:

```
void setup()
{
  tud_disconnect();      // don't let USB enumerate yet
  Serial1.begin(115200);
  stdout = funopen((void *)(&Serial1), NULL, myWrite, NULL, NULL);
  setlinebuf(stdout);
  printf("\n\rSerial1 initialized as stdout for printf\r\n");
  delay(20000);
  tud_connect();   // now we're ready to start USB
```