adafruit/Adafruit_TinyUSB_Arduino #103: Compilation fails on SAMD51 board without a Serial1 declaration

PaintYourDragon, 2021-07-07:
### Operating System

MacOS

### IDE version

Arduino 1.8.15

### Board

Adafruit Monster M4sk

### BSP version

1.7.3

### TinyUSB Library version

1.3.0 via Arduino Library Manager

### Sketch

Adafruit_Learning_System_Guides/M4_Eyes

### What happened ?

Project compilation fails with the following error:
/Users/pburgess/Documents/Arduino/libraries/Adafruit_TinyUSB_Library/src/arduino/ports/samd/Adafruit_TinyUSB_samd.cpp:65:3: error: 'Serial1' was not declared in this scope; did you mean 'Serial'?

**This board (Monster M4sk) does not have a Serial1 declaration in ArduinoCore_samd.**

Workaround is to edit tusb_config_samd.h and change line 49 to:
`#define CFG_TUSB_DEBUG 0`
Project then builds, uploads and runs successfully.

Perhaps this should be the default case when headed for the Arduino Library Manager? Or, if debug 2 is the default and intended case, maybe the board definition should get a Serial1 added, in which case I’ll check for any other SAMD51 boards without one.

### How to reproduce ?

Build project described above

hathach, 2021-07-07:
Opps, I am sorry, I think I leave the debugging option enabled by mistake. Intended to enable it along with debug menu option, manually enabled for testing and forgot to change it back. Revert it asap.

PaintYourDragon, 2021-07-07:
Looking through the history, this might have come up during a refactor of many files, so reverting might not be ideal. Maybe best to set debug to 0, bump the version number in library.properties and draft new release so Arduino Library Manager picks up the change?

hathach, 2021-07-07:
yeah, by reverting, I mean set it back to 0 :sweat_smile: PR is on the way #104, will merge and release asap.

PR #104 (2021-07-07): revert CFG_TUSB_DEBUG on samd port
fix #103
closed 2021-07-07 by hathach