adafruit/Adafruit_TinyUSB_Arduino #105: TinyUSB Mass Storage Device incompatibility with nRF52840

Finomnis, 2021-07-11:
### Operating System

Windows 10

### IDE version

Platformio Core 5.1.1

### Board

Feather nRF 52840 Express

### BSP version

Nordic nRF52 8.1.0

### TinyUSB Library version

1.3.2

### Sketch

```
#include "SPI.h"
#include "Adafruit_SPIFlash.h"
#include "Adafruit_TinyUSB.h"

int32_t msc_read_cb(uint32_t lba, void *buffer, uint32_t bufsize);
int32_t msc_write_cb(uint32_t lba, uint8_t *buffer, uint32_t bufsize);
void msc_flush_cb(void);

Adafruit_FlashTransport_QSPI flashTransport;
Adafruit_SPIFlash flash(&flashTransport);

// USB Mass Storage object
Adafruit_USBD_MSC usb_msc;

// the setup function runs once when you press reset or power the board
void setup()
{
    flash.begin();

    // Set disk vendor id, product id and revision with string up to 8, 16, 4 characters respectively
    usb_msc.setID("Adafruit", "External Flash", "1.0");

    // Set callback
    usb_msc.setReadWriteCallback(msc_read_cb, msc_write_cb, msc_flush_cb);

    // Set disk size, block size should be 512 regardless of spi flash page size
    usb_msc.setCapacity(flash.size() / 512, 512);

    // MSC is ready for read/write
    usb_msc.setUnitReady(true);

    usb_msc.begin();
}

void loop()
{
}

int32_t msc_read_cb(uint32_t lba, void *buffer, uint32_t bufsize)
{
    return flash.readBlocks(lba, (uint8_t *)buffer, bufsize / 512) ? bufsize : -1;
}

int32_t msc_write_cb(uint32_t lba, uint8_t *buffer, uint32_t bufsize)
{
    return flash.writeBlocks(lba, buffer, bufsize / 512) ? bufsize : -1;
}

void msc_flush_cb(void)
{
    flash.syncBlocks();
}
```

### What happened ?

It should compile. This is the absolute minimal version of a USB Mass Storage Device for the QSPI chip of the Adafruit Feather nRF52840 Express.

However, the build throws errors, saying that `Adafruit_USBD_MSC` is a purely virtual class, which it shouldn't be. Therefore I suspect a bug in the library.

### How to reproduce ?

- Create a new PlatformIO project for the `adafruit_feather_nrf52840` target.
- Install the following dependencies:
  - `adafruit/Adafruit SPIFlash`
  - `adafruit/Adafruit TinyUSB Library`
  - `adafruit/SdFat - Adafruit Fork`

The `platformio.ini` should then look like following:
```
[env:adafruit_feather_nrf52840]
platform = nordicnrf52
board = adafruit_feather_nrf52840
framework = arduino
lib_deps = 
	adafruit/Adafruit SPIFlash@^3.5.2
	adafruit/Adafruit TinyUSB Library@^1.3.2
	adafruit/SdFat - Adafruit Fork@^1.2.4
```

Then, copy the above code to `main.cpp` and try to build the project.

It should throw an error during the build process.

### Debug Log

```
Processing adafruit_feather_nrf52840 (platform: nordicnrf52; board: adafruit_feather_nrf52840; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/adafruit_feather_nrf52840.html
PLATFORM: Nordic nRF52 (8.1.0) > Adafruit Feather nRF52840 Express
HARDWARE: NRF52840 64MHz, 243KB RAM, 796KB Flash
DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-arduinoadafruitnrf52 1.2100.201028 (21.0)
 - tool-sreccat 1.164.0 (1.64)
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Adafruit SPIFlash> 3.5.2
|   |-- <SPI> 1.0
|   |-- <SdFat - Adafruit Fork> 1.2.4
|   |   |-- <SPI> 1.0
|-- <Adafruit TinyUSB Library> 1.3.2
|-- <SdFat - Adafruit Fork> 1.2.4
|   |-- <SPI> 1.0
|-- <SPI> 1.0
Building in release mode
Compiling .pio\build\adafruit_feather_nrf52840\src\main.cpp.o
Compiling .pio\build\adafruit_feather_nrf52840\lib83f\SPI\SPI.cpp.o
Compiling .pio\build\adafruit_feather_nrf52840\lib83f\SPI\SPI_nrf52832.cpp.o
Compiling .pio\build\adafruit_feather_nrf52840\lib49a\SdFat - Adafruit Fork\FatLib\FatFile.cpp.o
Compiling .pio\build\adafruit_feather_nrf52840\lib49a\SdFat - Adafruit Fork\FatLib\FatFileLFN.cpp.o
src\main.cpp:13:19: error: cannot declare variable 'usb_msc' to be of abstract type 'Adafruit_USBD_MSC'
 Adafruit_USBD_MSC usb_msc;
                   ^~~~~~~
In file included from .pio\libdeps\adafruit_feather_nrf52840\Adafruit TinyUSB Library\src/Adafruit_TinyUSB.h:44:0,
                 from src\main.cpp:3:
.pio\libdeps\adafruit_feather_nrf52840\Adafruit TinyUSB Library\src/arduino/msc/Adafruit_USBD_MSC.h:30:7: note:   because the following virtual functions are pure within 'Adafruit_USBD_MSC':
 class Adafruit_USBD_MSC : public Adafruit_USBD_Interface {
       ^~~~~~~~~~~~~~~~~
In file included from C:\Users\Martin\.platformio\packages\framework-arduinoadafruitnrf52\cores\nRF5\TinyUSB\Adafruit_TinyUSB_ArduinoCore/Adafruit_TinyUSB_Core.h:35:0,
                 from C:\Users\Martin\.platformio\packages\framework-arduinoadafruitnrf52\cores\nRF5/Arduino.h:72,
                 from C:\Users\Martin\.platformio\packages\framework-arduinoadafruitnrf52\libraries\SPI/SPI.h:24,
                 from src\main.cpp:1:
C:\Users\Martin\.platformio\packages\framework-arduinoadafruitnrf52\cores\nRF5\TinyUSB\Adafruit_TinyUSB_ArduinoCore/Adafruit_USBD_Device.h:33:22: note: 
virtual uint16_t Adafruit_USBD_Interface::getDescriptor(uint8_t, uint8_t*, uint16_t)
     virtual uint16_t getDescriptor(uint8_t itfnum, uint8_t* buf, uint16_t bufsize) = 0;
                      ^~~~~~~~~~~~~
In file included from C:\Users\Martin\.platformio\packages\framework-arduinoadafruitnrf52\libraries\SPI\SPI.h:24:0,
                 from C:\Users\Martin\.platformio\packages\framework-arduinoadafruitnrf52\libraries\SPI\SPI.cpp:26:
C:\Users\Martin\.platformio\packages\framework-arduinoadafruitnrf52\cores\nRF5/Arduino.h: In instantiation of 'decltype (((b < a) ? b :  a)) min(const T&, const L&) [with T = unsigned int; L = int; decltype (((b < a) ? b :  a)) = unsigned int]':
C:\Users\Martin\.platformio\packages\framework-arduinoadafruitnrf52\libraries\SPI\SPI.cpp:202:50:   required from here
C:\Users\Martin\.platformio\packages\framework-arduinoadafruitnrf52\cores\nRF5/Arduino.h:93:15: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     return (b < a) ? b : a;
            ~~~^~~~
*** [.pio\build\adafruit_feather_nrf52840\src\main.cpp.o] Error 1
============================================================== [FAILED] Took 6.66 seconds ==============================================================
```

ladyada, 2021-07-11:
please try arduino IDE to verify it isnt a bug in platformio

Finomnis, 2021-07-11:
@ladyada As stated, it compiles and runs without a problem in the Arduino IDE.
If this is a but in PlatformIO, where should I report it? I'd switch to Arduino IDE, but Arduino IDE is just very limited with larger projects.

ladyada, 2021-07-11:
yes then sounds like it is a bug in platformio - they have tech support for their product elsewhere = thank you :)

closed 2021-07-11 by ladyada

Finomnis, 2021-07-11:
Thanks for your time anyway :)

ladyada, 2021-07-11:
good luck!