adafruit/Adafruit_TinyUSB_Arduino #106: Metro M4 platformio no USB functionality with tinyUSB

laermansjohan, 2021-07-12:
### Operating System

Linux

### IDE version

Platformio with CLion

### Board

Metro M4

### BSP version

Adafruit SAMD Boards version=1.7.2

### TinyUSB Library version

1.3.2 - 2021.07.07

### Sketch

src/main.ccp code:

```
#include <Arduino.h>

#include "SPI.h"
#include "SdFat.h"
#include "Adafruit_SPIFlash.h"
#include "Adafruit_TinyUSB.h"

#if defined(EXTERNAL_FLASH_USE_QSPI)
Adafruit_FlashTransport_QSPI flashTransport;

#elif defined(EXTERNAL_FLASH_USE_SPI)
Adafruit_FlashTransport_SPI flashTransport(EXTERNAL_FLASH_USE_CS, EXTERNAL_FLASH_USE_SPI);

  #else
    #error No QSPI/SPI flash are defined on your board variant.h !
#endif

Adafruit_SPIFlash flash(&flashTransport);

// file system object from SdFat
FatFileSystem fatfs;

FatFile root;
FatFile file;

// USB Mass Storage object
Adafruit_USBD_MSC usb_msc;

// Set to true when PC write to flash
bool changed;
// Callback invoked when received READ10 command.
// Copy disk's data to buffer (up to bufsize) and
// return number of copied bytes (must be multiple of block size)
int32_t msc_read_cb (uint32_t lba, void* buffer, uint32_t bufsize)
{
    // Note: SPIFLash Bock API: readBlocks/writeBlocks/syncBlocks
    // already include 4K sector caching internally. We don't need to cache it, yahhhh!!
    return flash.readBlocks(lba, (uint8_t*) buffer, bufsize/512) ? bufsize : -1;
}

// Callback invoked when received WRITE10 command.
// Process data in buffer to disk's storage and
// return number of written bytes (must be multiple of block size)
int32_t msc_write_cb (uint32_t lba, uint8_t* buffer, uint32_t bufsize)
{
    digitalWrite(3, HIGH);

    // Note: SPIFLash Bock API: readBlocks/writeBlocks/syncBlocks
    // already include 4K sector caching internally. We don't need to cache it, yahhhh!!
    return flash.writeBlocks(lba, buffer, bufsize/512) ? bufsize : -1;
}

// Callback invoked when WRITE10 command is completed (status received and accepted by host).
// used to flush any pending cache.
void msc_flush_cb (void)
{
    // sync with flash
    flash.syncBlocks();

    // clear file system's cache to force refresh
    fatfs.cacheClear();

    changed = true;

    digitalWrite(3, LOW);
}
// the setup function runs once when you press reset or power the board
void setup()
{
    pinMode(3, OUTPUT);

    flash.begin();

    // Set disk vendor id, product id and revision with string up to 8, 16, 4 characters respectively
    usb_msc.setID("Adafruit", "External Flash", "1.0");

    // Set callback
    usb_msc.setReadWriteCallback(msc_read_cb, msc_write_cb, msc_flush_cb);

    // Set disk size, block size should be 512 regardless of spi flash page size
    usb_msc.setCapacity(flash.size()/512, 512);

    // MSC is ready for read/write
    usb_msc.setUnitReady(true);

    usb_msc.begin();

    // Init file system on the flash
    fatfs.begin(&flash);

    Serial.begin(115200);
    //while ( !Serial ) delay(10);   // wait for native usb

    Serial.println("Adafruit TinyUSB Mass Storage External Flash example");
//    Serial.print("JEDEC ID: "); Serial.println(flash.getJEDECID(), HEX);
//    Serial.print("Flash size: "); Serial.println(flash.size());

    changed = true; // to print contents initially
}

void loop()
{
    if ( changed )
    {
        changed = false;

        if ( !root.open("/") )
        {
            Serial.println("open root failed");
            return;
        }

        Serial.println("Flash contents:");

        // Open next file in root.
        // Warning, openNext starts at the current directory position
        // so a rewind of the directory may be required.
        while ( file.openNext(&root, O_RDONLY) )
        {
            file.printFileSize(&Serial);
            Serial.write(' ');
            file.printName(&Serial);
            if ( file.isDir() )
            {
                // Indicate a directory.
                Serial.write('/');
            }
            Serial.println();
            file.close();
        }

        root.close();

        Serial.println();
        delay(1000); // refresh every 0.5 second
    }
}
```
platform.ini:

```
[env:adafruit_metro_m4]
platform = atmelsam
board = adafruit_metro_m4
framework = arduino
build_flags = -DUSE_TINYUSB
lib_deps =
    https://github.com/adafruit/SdFat
    https://github.com/adafruit/Adafruit_SPIFlash
```

### What happened ?

I have a Metro M4 board where the "msc_external_flash" sketch worked (about one year ago) with the Adafruit tinyUSB  on platformio and CLion.
After updating pio and the libraries, the board USB functions stopped working.(from tinyUSB 0.10.0, Boards version=1.5.14 -> tinyUSB 1.3.2, Boards version=1.7.2, pio 5.1.1 )

### How to reproduce ?

I have no compilation errors, however all USB functions of the board do not work, no working msc external flash, nor the serial port that is working. The Tx & Rx LEDs on the Metro stay lid. Other code in the loop runs, the board is not locked.

When I do not set the  `build_flags = -DUSE_TINYUSB` in the platformio.ini file and do not use usb_msc (no definition of Adafruit_TinyUSB.h) the boards USB initializes the right way and the serial port works. Also the Tx & Rx LEDs work correct in that case.

Is there something that I overlooked, do I miss some build flags in the platform.ini file? Thanks in advance for looking in to it.

### Debug Log

Just to show the versions used:

PLATFORM: Atmel SAM (6.2.0)   Adafruit Metro M4
HARDWARE: SAMD51J19A 120MHz, 192KB RAM, 512KB Flash
DEBUG: Current (atmel-ice) External (atmel-ice, jlink)
PACKAGES: 
 - framework-arduino-samd-adafruit 1.7.2 
 - framework-cmsis 2.50400.181126 (5.4.0) 
 - framework-cmsis-atmel 1.2.2 
 - toolchain-gccarmnoneeabi 1.90301.200702 (9.3.1)
LDF: Library Dependency Finder -  http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 35 compatible libraries
Scanning dependencies...
Dependency Graph
|--  SdFat - Adafruit Fork  1.2.4+sha.68dd2d4
|   |--  SPI  1.0
|   |   |--  Adafruit Zero DMA Library  1.1.0
|   |   |   |--  Adafruit TinyUSB Library  1.3.2
|   |   |--  Adafruit TinyUSB Library  1.3.2
|   |--  Adafruit Zero DMA Library  1.1.0
|   |   |--  Adafruit TinyUSB Library  1.3.2
|--  Adafruit SPIFlash  3.5.2+sha.fa5d4c7
|   |--  SPI  1.0
|   |   |--  Adafruit Zero DMA Library  1.1.0
|   |   |   |--  Adafruit TinyUSB Library  1.3.2
|   |   |--  Adafruit TinyUSB Library  1.3.2
|   |--  Adafruit Zero DMA Library  1.1.0
|   |   |--  Adafruit TinyUSB Library  1.3.2
|   |--  SdFat - Adafruit Fork  1.2.4+sha.68dd2d4
|   |   |--  SPI  1.0
|   |   |   |--  Adafruit Zero DMA Library  1.1.0
|   |   |   |   |--  Adafruit TinyUSB Library  1.3.2
|   |   |   |--  Adafruit TinyUSB Library  1.3.2
|   |   |--  Adafruit Zero DMA Library  1.1.0
|   |   |   |--  Adafruit TinyUSB Library  1.3.2
|--  Adafruit TinyUSB Library  1.3.2
|--  SPI  1.0
|   |--  Adafruit Zero DMA Library  1.1.0
|   |   |--  Adafruit TinyUSB Library  1.3.2
|   |--  Adafruit TinyUSB Library  1.3.2

hathach, 2021-07-12:
First of all, If the code is stock example, you could just simply name it examples/msc/etc.. otherwise please attached it as file. Long issue description make it difficult for follow.

For your actual question, I have no clues since I don't use platformio, maybe you should ask on their forum.

hathach, 2021-07-16:
Maybe this could help https://community.platformio.org/t/tinyusb-featherwing-nrf52840-incompatibility/22434

closed 2021-07-16 by hathach

laermansjohan, 2021-07-20:
Hi,

Thanks for the propositions. Via the link i came across the same problem. Check [link](https://github.com/platformio/platform-atmelsam/issues/164) for the solution that worked for me.

Johan