adafruit/Adafruit_TinyUSB_Arduino #107: HardFault hanling bus reset (race condition on reset?)

eugeneanikin, 2021-07-13:
### Operating System

MacOS

### IDE version

Arduino 1.8.15

### Board

my nrf52840 board, likely all other

### BSP version

github latest

### TinyUSB Library version

github latest

### Sketch

Blink with  Serial.begin(115200);

### What happened ?

Burn bootloader with jlink, upload sketch via serial. Sometime the board freezes and the sketch doesn't run, serial port disappears.

### How to reproduce ?

See above.

Stack trace:
#0  HardFault_Handler ()
    at /System/Volumes/Data/qnap/Public/Eugene/Arduino/hardware/Adafruit/Adafruit_nRF52_Arduino/
cores/nRF5/linker/gcc_startup_nrf52840.S:313
#1  <signal handler called>
#2  0x0002d540 in memcpy ()
#3  0x0002b1bc in prvCopyDataToQueue (xPosition=0, pvItemToQueue=0x2003ff5c, pxQueue=0x0)
    at /System/Volumes/Data/qnap/Public/Eugene/Arduino/hardware/Adafruit/Adafruit_nRF52_Arduino/
cores/nRF5/freertos/Source/queue.c:2068
#4  xQueueGenericSendFromISR (xQueue=0x0, pvItemToQueue=pvItemToQueue@entry=0x2003ff7c,
    pxHigherPriorityTaskWoken=pxHigherPriorityTaskWoken@entry=0x2003ff78,
    xCopyPosition=xCopyPosition@entry=0)
    at /System/Volumes/Data/qnap/Public/Eugene/Arduino/hardware/Adafruit/Adafruit_nRF52_Arduino/
cores/nRF5/freertos/Source/queue.c:967
#5  0x00029064 in osal_queue_send (in_isr=true, data=0x2003ff7c, qhdl=<optimized out>)
    at /System/Volumes/Data/qnap/Public/Eugene/Arduino/hardware/Adafruit/Adafruit_nRF52_Arduino/
libraries/Adafruit_TinyUSB_Arduino/src/osal/osal_freertos.h:151
#6  dcd_event_handler (in_isr=true, event=0x2003ff7c)
    at /System/Volumes/Data/qnap/Public/Eugene/Arduino/hardware/Adafruit/Adafruit_nRF52_Arduino/
libraries/Adafruit_TinyUSB_Arduino/src/device/usbd.c:1067
#7  dcd_event_bus_reset (rhport=rhport@entry=0 '\000', speed=speed@entry=TUSB_SPEED_FULL,
    in_isr=in_isr@entry=true)
    at /System/Volumes/Data/qnap/Public/Eugene/Arduino/hardware/Adafruit/Adafruit_nRF52_Arduino/
libraries/Adafruit_TinyUSB_Arduino/src/device/usbd.c:1082
#8  0x0002a5a8 in dcd_int_handler (rhport=<optimized out>)
    at /System/Volumes/Data/qnap/Public/Eugene/Arduino/hardware/Adafruit/Adafruit_nRF52_Arduino/
libraries/Adafruit_TinyUSB_Arduino/src/portable/nordic/nrf5x/dcd_nrf5x.c:548
#9  <signal handler called>
#10 0x0002cbf2 in vPortStartFirstTask ()
    at /System/Volumes/Data/qnap/Public/Eugene/Arduino/hardware/Adafruit/Adafruit_nRF52_Arduino/
cores/nRF5/freertos/portable/GCC/nrf52/port.c:53
#11 0x0002c932 in xPortStartScheduler ()
    at /System/Volumes/Data/qnap/Public/Eugene/Arduino/hardware/Adafruit/Adafruit_nRF52_Arduino/
cores/nRF5/freertos/portable/CMSIS/nrf52/port_cmsis.c:240
#12 0x00000000 in ?? ()

Notice xQueue is NULL. It appears to me the interrupt arrived before tud_init() had a chance to create the queue.   
Should we just ignore these early events?

hathach, 2021-07-26:
I couldn't reproduce the issue, but it is indeed legit. I have update the implementation to move part of the usb hardware init after tus b init with pr #112 . Would you mind trying it out to see if this fixes the issue.

eugeneanikin, 2021-07-26:
Tried it. Seems allright. You never know with race conditions. It was probably originally triggered by a faulty bootloader I was using. I changed it since then, so I can't reproduce this bug either.

hathach, 2021-07-26:
I see your bootloader probably doesn't disable USB IRQ or didn't power off the peripherals before jumping to app or thing like that. Normally the vbus sequence take much longer to stable and bus reset take a few ms afterwards to trigger IRQ. Anyway it is still a race, fixing is still good thing to do.

PR #112 (2021-07-26): nrf5x fix race condition
 where init usb hardware cause USB IRQ to trigger before tusb is inited causing hardfault since usb control state (queue, task) is not inited. Should fix #107
closed 2021-07-26 by hathach