adafruit/Adafruit_TinyUSB_Arduino #115: Sluggish USB Serial behaviour when accessed through the Stream class from another class

CreativeRobotics, 2021-08-04:
### Operating System

Windows 10

### IDE version

Arduino 1.8.13

### Board

Adafruit ItsyBitsy M4

### BSP version

1.7.3

### TinyUSB Library version

Latest release in Adafruit SAMD boards

### Sketch

[BasicCommands_VersionTest_TinyUSB_VERSIONB.zip](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/6932321/BasicCommands_VersionTest_TinyUSB_VERSIONB.zip)
[BasicCommands_VersionTest_TinyUSB.zip](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/6932322/BasicCommands_VersionTest_TinyUSB.zip)

### What happened ?

I updated all my libraries and the Adafruit board distribution and tinyUSB now appear to be only semi functional for USB serial devices under certain conditions.

I'm using a library called Commander to provide an on-device command line interface. When I use the Arduino USB stack Commander works fine, but when I use the tinyUSB stack it is very noticeably slower. This issue has only appeared since I updated the Adafruit SAMD core and tinyUSB library.

I have tried two versions of Commander, the latest release that inherits from the Stream class, and an older version that doesn't. Both are slow, but the latest version is crazy slow. Printing a few lines of text takes several seconds. Again, this issue only appeared when I updated Adafruits SAMD core and tiny USB library, and is not present when using the Arduino USB core.

Before I updated my libraries and the Adafruit board dist to the latest versions, I was not having this problem at all with tinyUSB. I was using Commander in a sketch that worked as a USB flash drive and USB serial port and both USB interfaces worked side by side without problems. This sketch is now totally non functional.

I've included links to two test files that show the differences in speed between the Arduino and tinyUSB stacks when using my library to respond to commands.

I've tried this on adafruit boards with an M0+ and M4 mcu.

### How to reproduce ?

Download both the zip files.
BasicCommands_VersionTest_TinyUSB contains a sketch using an old version of Commander.
Compile this for an adafruit samd board using the tinyUSB stack.
Open a serial port and type 'help.
A help page will print out and will take just under a second.
Comment out #include "Adafruit_TinyUSB.h" and recompile using the Arduino USB stack.
Open a serial port and type 'help.
A help page will print out fast enough to appear instantly.

BasicCommands_VersionTest_TinyUSB_VERSIONB contains a sketch using the latest version of Commander that inherits the Stream object.
Compile this for an adafruit samd board using the tinyUSB stack.
Open a serial port and type 'help.
A help page will print out and will take several seconds.
Comment out #include "Adafruit_TinyUSB.h" and recompile using the Arduino USB stack.
Open a serial port and type 'help.
A help page will print out fast enough to appear instantly.

CreativeRobotics, 2021-08-04:
[TinyUSBClassTest.zip](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/6933570/TinyUSBClassTest.zip)

Just tested and its related to the use of yield(). The attached sketch just has a simple class that inherits Stream, and prints a message. If you include yield() in the implimentation of write() then the tinyUSB stacks print output is very slow, but its super fast with the Arduino stack.

CreativeRobotics, 2021-08-05:
Ok, after sleeping on it I can see how calling yield() for every byte sent might cause an issue in my library. Close this if you think its not an issue, but consider that there may be something in tinyUSB that is consuming more time than needed when yield is called.

hathach, 2021-08-05:
`yield()` is updated to also call Serial.flush() internally since it assume loop() would give up mcu control https://github.com/adafruit/ArduinoCore-samd/blob/master/cores/arduino/delay.c#L126. Because you are writing 1 char then flush, it will indeed slow down the whole transfer. Arduino stack handle all usb code within ISR and does not actual flush when yield(). 

It works great with most of user scenario, it may not perfect but there is no 1 implementation fit all. Out of curiosity, why would you yield() every of the Stream API, it looks a bit odd, could you alter your implementation. Since normally for stream, it is buffered either until full, special char `\n` or manually flush().

CreativeRobotics, 2021-08-06:
Yes, I was being stupid (doubly so because I was inheriting Print, not Stream)- I hadn't appreciated how yield() was being called on every character written, it was code carried over from the version that didn't inherit from print and was calling yield in my own implimentation of print and println, but not in write().

The reason it was in there in the first place was as a work around when using it on ESP32 devices, sending large amounts of data over a bluetooth serial connection was causing issues and yield() fixed the problem.

Thanks for pointing out how delay() calls yield() - That actually helps make my code simpler as the library also has a option to introduce a delay after n bytes are sent, so I can use this to ensure yield is called periodically only when the user needs it.

closed 2021-08-06 by CreativeRobotics

hathach, 2021-08-06:
not a problem at all, these behind-the-scene code can be confusing, different from one platform to another and and not very obvious to user as well. I am glad you figure it out.