adafruit/Adafruit_TinyUSB_Arduino #125: nRF52840 - USB HID doesn't work if Serial.begin and usb_hid.begin are sequenced incorrectly

jpconstantineau, 2021-08-21:
### Operating System

Windows 10

### IDE version

1.8.15

### Board

Feather nRF52840 Express

### BSP version

0.24.0

### TinyUSB Library version

version included with BSP 0.24.0

### Sketch

/*********************************************************************
 Adafruit invests time and resources providing this open source code,
 please support Adafruit and open-source hardware by purchasing
 products from Adafruit!

 MIT license, check LICENSE for more information
 Copyright (c) 2019 Ha Thach for Adafruit Industries
 All text above, and the splash screen below must be included in
 any redistribution
*********************************************************************/

// this is from examples/Adafruit TinyUSB library/HID/hid_composite

#include "Adafruit_TinyUSB.h"

/* This sketch demonstrates multiple report USB HID.
 * Press button pin will move
 * - mouse toward bottom right of monitor
 * - send 'a' key
 *
 * Depending on the board, the button pin
 * and its active state (when pressed) are different
 */
#if defined ARDUINO_SAMD_CIRCUITPLAYGROUND_EXPRESS
  const int pin = 4; // Left Button
  bool activeState = true;
#elif defined ARDUINO_NRF52840_FEATHER
  const int pin = 7; // UserSw
  bool activeState = false;
#else
  const int pin = 12;
  bool activeState = false;
#endif

// Report ID
enum
{
  RID_KEYBOARD = 1,
  RID_MOUSE,
  RID_CONSUMER_CONTROL, // Media, volume etc ..
};

// HID report descriptor using TinyUSB's template
uint8_t const desc_hid_report[] =
{
  TUD_HID_REPORT_DESC_KEYBOARD( HID_REPORT_ID(RID_KEYBOARD) ),
  TUD_HID_REPORT_DESC_MOUSE   ( HID_REPORT_ID(RID_MOUSE) ),
  TUD_HID_REPORT_DESC_CONSUMER( HID_REPORT_ID(RID_CONSUMER_CONTROL) )
};

// USB HID object
Adafruit_USBD_HID usb_hid;

// the setup function runs once when you press reset or power the board
void setup()
{

// Note that Serial.begin was moved to the top of setup  to allow for debugging.

  Serial.begin(115200);
  while ( !Serial ) delay(10);   // for nrf52840 with native usb
  Serial.println("Adafruit TinyUSB HID Composite example");
  
  usb_hid.setPollInterval(2);
  usb_hid.setReportDescriptor(desc_hid_report, sizeof(desc_hid_report));
  //usb_hid.setStringDescriptor("TinyUSB HID Composite");

  usb_hid.begin();

  // Set up button, pullup opposite to active state
  pinMode(pin, activeState ? INPUT_PULLDOWN : INPUT_PULLUP);

  // wait until device mounted
  while( !USBDevice.mounted() ) delay(1);
}

void loop()
{
  // poll gpio once each 10 ms
  delay(10);

  // Whether button is pressed
  bool btn_pressed = (digitalRead(pin) == activeState);

  // Remote wakeup
  if ( USBDevice.suspended() && btn_pressed )
  {
    // Wake up host if we are in suspend mode
    // and REMOTE_WAKEUP feature is enabled by host
    USBDevice.remoteWakeup();
  }

  /*------------- Mouse -------------*/
  if ( usb_hid.ready() && btn_pressed )
  {
    int8_t const delta = 5;
    usb_hid.mouseMove(RID_MOUSE, delta, delta); // right + down

    // delay a bit before attempt to send keyboard report
    delay(10);
  }

  /*------------- Keyboard -------------*/
  if ( usb_hid.ready() )
  {
    // use to send key release report
    static bool has_key = false;

    if ( btn_pressed )
    {
      uint8_t keycode[6] = { 0 };
      keycode[0] = HID_KEY_A;

      usb_hid.keyboardReport(RID_KEYBOARD, 0, keycode);

      has_key = true;
    }else
    {
      // send empty key report if previously has key pressed
      if (has_key) usb_hid.keyboardRelease(RID_KEYBOARD);
      has_key = false;
    }

    // delay a bit before attempt to send consumer report
    delay(10);
  }

  /*------------- Consumer Control -------------*/
  if ( usb_hid.ready() )
  {
    // Consumer Control is used to control Media playback, Volume, Brightness etc ...
    // Consumer report is 2-byte containing the control code of the key
    // For list of control check out https://github.com/hathach/tinyusb/blob/master/src/class/hid/hid.h#L544

    // use to send consumer release report
    static bool has_consumer_key = false;

    if ( btn_pressed )
    {
      // send volume down (0x00EA)
      usb_hid.sendReport16(RID_CONSUMER_CONTROL, HID_USAGE_CONSUMER_VOLUME_DECREMENT);
      has_consumer_key = true;
    }else
    {
      // release the consume key by sending zero (0x0000)
      if (has_consumer_key) usb_hid.sendReport16(RID_CONSUMER_CONTROL, 0);
      has_consumer_key = false;
    }
  }
}

### What happened ?

If serial.begin is located before usb_hid.begin, usb_hid.ready() never returns true. (HID doesn't work)

If serial.begin is located after usb_hid.begin, usb_hid.ready() works properly

### How to reproduce ?

load sketch above - it doesn't work.
Move serial.begin down at the end of setup and it works.

### Debug Log

no logs...  just press the user button on the feather and "a" should be typed and it doesn't...

### Screenshots

n/a

jpconstantineau, 2021-08-21:
Not sure if this should have been filed with TinyUSB, here or the Adafruit_nRF52_Arduino repo...  Here made more sense.

hathach, 2021-08-22:
Please post sketch as attached file for readability

jpconstantineau, 2021-08-22:
Ok will do.  
Some more updates... Initializing BLE before USB also seem to prevent USB HID from working.

jpconstantineau, 2021-08-22:
[hid_composite_bug.ino.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/7027516/hid_composite_bug.ino.txt)

Find attached sketch (recreated from hid_composite.ino) but renamed with .txt extension as GH doesn't support .ino extensions.

todbot, 2021-08-22:
What happens if you comment out the “while(!Serial)”?

jpconstantineau, 2021-08-22:
removing it seems to resolve things.  Even a 1 or 10 second delay, it works.
I am stumped as last night I had lots of problems with it.

hathach, 2021-08-23:
this has been asked and answered multiple time, check out those old issue. I make a sum up here. The delay mean device wait until the usb enumeration is complete (plus Serial is connected), at that time the host and device is done with negotiate device driver. Adding more driver requires an detach and retach.

```C
while ( !Serial ) delay(10); // for nrf52840 with native usb
```

closed 2021-08-23 by hathach

jpconstantineau, 2021-08-23:
Thanks for the explanation.  Makes sense now.
If it's been asked multiple times, I suspect some comments pointing to documentation might help. How that would look, not too sure considering that it's a library.  Perhaps a comment in the examples.

As a start, perhaps there is introduction documentation somewhere that could summarize/explain how USB works and link it here.  Where I struggle is the meaning of mounted, (not) suspended, and ready.  I am sure there are lots of web pages/documentation out there from various manufacturers but perhaps something you know that's a good reference would be useful.

Again, thanks for the clarification.