adafruit/Adafruit_TinyUSB_Arduino #127: Linux CDC (/dev/ttyACM*) <=> TinyUSB on nRF52 freezes with too much data

egnor, 2021-10-01:
### Operating System

Linux

### IDE version

ardi v2.4.1 (based on a snapshot of arduino-cli I believe)

### Board

Adafruit Circuit Playground "Bluefruit" nRF52840

### BSP version

Adafruit nRF52 1.0.0

### TinyUSB Library version

Adafruit TinyUSB Library 1.4.4 (and whichever tinyusb that bundles)

### Sketch

There's a lot going on here, it it's useful I can work on a MUCH more targeted repro example.

[bluefruit.ino.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/7266071/bluefruit.ino.txt)

### What happened ?

First of all many thanks :pray: :tada: for excellent and very useful software!

When writing data rapidly _from_ Linux host _to_ the board ("OUT" as USB calls it), that direction will freeze. Data will still flow from device to host, but attempts to send data from Linux hang (in fact even attempting to close the port hangs once some write have happened). The only recourse seems to be unplug/replug the device. I might expect buffer overruns under these conditions but certainly don't expect a permanently borked port.

I took some usbmon traces which I can share if they are interesting, they seem to show the device failing to respond to requests for data in that direction?? But I am _not_ very versed in the nuts and bolts of USB communication.

POSSIBLY relevant: https://github.com/adafruit/Adafruit_TinyUSB_Arduino/issues/57 (but the opposite direction).

This reproduces on a Raspberry Pi running Raspberry Pi OS (buster, fully updated) and also on a more or less ordinary Intel Linux machine running Ubuntu 21.04. I have a "spammer" script on the Linux side that can generally cause this to happen within a few seconds on any host I've tried.

I know that as it stands this is not a great bug report but I can certainly collect more info and build repro cases. Hopefully the fact that it's happening on a widely available Adafruit board is helpful, just let me know what you'd need.

### How to reproduce ?

Run a sketch on the device that consumes data. Send a lot of data to the device (perhaps more quickly than it can read).

### Debug Log

(I'm not sure how to collect this log, but I can research it.)

hathach, 2021-10-01:
thank you for opening the issue, however your sketch is too complicated, which can have its own issue e.g ble interfering with usb etc.. Can you 
1. Make an simplest sketch as possible with usb only e.g Serial echo or similar to narrow down the investigated code
2. Please update the **How to reproduce** to be precise, `what do you do to send lots of data`. If you copy and paste some texts, please put the text here so that I could do it as well. Or if you send it via echo command please put the command here.
3. `ardi v2.4.1 (based on a snapshot of arduino-cli I believe)` I only use Arduino IDE please make sure you test with the Arduino IDE to rule out any 3rd party issue

egnor, 2021-10-01:
Okay! I wrote a mini serial-echo sketch, and a mini load-tester script, and have edited the first post to include the new repro case. Any help is welcome!! :bow:

egnor, 2021-10-01:
Reproduced with stock Arduino IDE.

egnor, 2021-10-01:
Reproduced (exact same behavior) with a Mac as the host.

egnor, 2021-10-01:
Reproduced (exact same behavior) on Feather nRF52840 (not surprising, it's a very similar design as the Circuit Playground, just doesn't have all the extra playground peripherals).

hathach, 2021-10-02:
@egnor superb ! thank you for your update, this is exactly what needed to reproduce the issue. I will try to test it out and post update here when available

hathach, 2021-11-12:
@egnor sorry for late response, I was busy with other works. I confirmed the issue occurred with your sketch and script. It is probably be the race condition or mutex locking issue. I will spend more time looking at this and post the update here. 

PS: I also realize that the Serial is lacking read(buf, bufsize) API. When adding the API to read multiple bytes at once, the hanged up is less likely to happen. So it is definitely a race condition

```C
void loop() {
  if (Serial.available()) {
    uint8_t buf[64];
    uint8_t count = Serial.read(buf, 64);
    Serial.write(buf, count);
    digitalToggle(LED_BUILTIN);
  }
}
```

hathach, 2021-11-13:
Update I put more effort looking debugging the nrf52. nrf52840 is not hanged, it appear to be running just fine:
- internal fifo of cdc indicate there is no data
- We could still put it into DFU mode with touch1200, which use control endpoint
- However, my linux (ubuntu) doesn't seem to be able to communicate with the port anymore, CDC Data endpoint

further troubleshooting

egnor, 2021-11-22:
That matches my understanding -- the nrf52840 is still running, but somehow USB-serial has wedged.

Glad to hear there's progress, let me know if I can help!

hathach, 2021-11-23:
@egnor this takes more time than troubleshoot than I would expect. Please give it a try

- #135  

I am glad it is fixed now. This has been 3rd time I got to fix the easydma race condition with nrf, previous one lack the intense test script like yours.

PR #135 (2021-11-26): fix nrf easy dma race condition
this fix #127
closed 2021-11-26 by hathach