adafruit/Adafruit_TinyUSB_Arduino #136: Compile Error for ESP32-S2: error: 'USB' was not declared in this scope

jockm, 2021-11-23:
### Operating System

Windows 10

### IDE version

Arduino 1.8.13

### Board

ESP32S2 Dev Module

### BSP version

2.0.1

### TinyUSB Library version

Release

### Sketch

examples/hid/hid_generic_inout and others, I just haven't exhaustively tested them all

### What happened ?

```
C:\Users\jock\OneDrive\Documents\Arduino\libraries\Adafruit_TinyUSB_Library\src\arduino\webusb\Adafruit_USBD_WebUSB.cpp: In constructor 'Adafruit_USBD_WebUSB::Adafruit_USBD_WebUSB(const void*)':

C:\Users\jock\OneDrive\Documents\Arduino\libraries\Adafruit_TinyUSB_Library\src\arduino\webusb\Adafruit_USBD_WebUSB.cpp:152:3: error: 'USB' was not declared in this scope

   USB.usbVersion(0x0210);

   ^~~

Using library Adafruit_TinyUSB_Library at version 1.6.0 in folder: C:\Users\jock\OneDrive\Documents\Arduino\libraries\Adafruit_TinyUSB_Library 

exit status 1

Error compiling for board ESP32S2 Dev Module.
```

### How to reproduce ?

1) Load hid_generic_inout sketch from examples
1) Set board to ESP32S2 Dev Module
1) Click on compile or upload

### Debug Log

N/A

### Screenshots

N/A

hathach, 2021-11-23:
try to set CDC enabled, MSC and DFU disabled via IDE menu. #133 . You could try to select one of Adafruit board to mimics its default config.

jockm, 2021-11-23:
@hathach Thank you! It does compile and upload, but I no longer have serial output on uart0.  Is this normal?

hathach, 2021-11-23:
you can use another menu option to map Serial to either USB CDC or HWUART. All of these options are designed by arduino-esp team. For further question, you could ask in espressif forum

closed 2021-11-23 by hathach

jockm, 2021-11-23:
@hathach Thank you, to clarify I wasn't asking you to debug it, but to clarify that was the expected behavior.  It is non, obvious that I should make these changes as I cannot find any documentation of it in this repository.  

Additionally it would probably be best if the code checked `ARDUINO_USB_CDC_ON_BOOT`, `ARDUINO_USB_MSC_ON_BOOT`, and `ARDUINO_USB_DFU_ON_BOOT` to produce an intentional `#error` instead of an actual compiler error if this is a known requirement.

Please don't think me ungrateful, but this would avoid annoying questions like this

jockm, 2021-11-23:
> you can use another menu option to map Serial to either USB CDC or HWUART. All of these options are designed by arduino-esp team. For further question, you could ask in espressif forum

Now to be clear I am NOT asking you to debug this, but that option does not appear in the 2.0.1 arduino-esp dev (the only branch with S2 support) definition for ESP32S2 Dev Module.  So you don't suggest that and cause confusion

hathach, 2021-11-24:
> Additionally it would probably be best if the code checked ARDUINO_USB_CDC_ON_BOOT, ARDUINO_USB_MSC_ON_BOOT, and ARDUINO_USB_DFU_ON_BOOT to produce an intentional #error instead of an actual compiler error if this is a known requirement.

This is a good suggestion, this library is added before release of 2.0.0. all the USB from arduino-esp32 is newly added, the reason for enabling CDC is to tell the arduino core to initialize usb stack. However, you will still be able to manually init it with USB.begin() (or something similar) in setup(). I will put a note on the readme.md regarding the esp32 usage.

> Please don't think me ungrateful, but this would avoid annoying questions like this
> Now to be clear I am NOT asking you to debug this, but that option does not appear in the 2.0.1 arduino-esp dev (the only branch with S2 support) definition for ESP32S2 Dev Module. So you don't suggest that and cause confusion

No I am not, we are totally cool. These menu option are added by espressif team, we have no influence on this. In fact, this library must be heavily modified to adapt to arduino-esp. They understand their implementation better and you will find better explanation on their support forum.

PS Menu option DFU and MSC on-boot have some conflict with this library, I haven't got time to get they co-exist and also want to wait for espressif to fully implement these feature to get the bigger picture. For now DFU/MSC on the menu should be disabled.