adafruit/Adafruit_TinyUSB_Arduino #140: cannot declare variable 'usb_web'

ratheeshpv, 2021-11-24:
### Operating System

Windows 7

### IDE version

Arduino 1.8.13

### Board

Seeeduino XIAO

### BSP version

couldn't find

### TinyUSB Library version

1.6.0

### Sketch

examples\WebUSB\webusb_serial

### What happened ?

Installed Adafruit_TinyUSB_Arduino-master.zip by using 'Add .ZIP Library'
Selected the example, 'webusb_serial'
clicked 'Upload' button
it gives following error,
"cannot declare variable 'usb_web' to be of abstract type 'Adafruit_USBD_WebUSB'"

### How to reproduce ?

1. Open Adafruit_TinyUSB_Arduino-master\examples\WebUSB\webusb_serial
2. Select board "Seeeduino XIAO"
3. Select USB stack, 'Tiny USB'
4. Select Port (for me, COM9 (Seeeduino XIAO))
5. click 'Upload'
6. gives error, "cannot declare variable 'usb_web' to be of abstract type 'Adafruit_USBD_WebUSB'  "

### Debug Log

In file included from C:\Users\sanujadithya\Documents\Arduino\libraries\Adafruit_TinyUSB_Arduino-master\src/Adafruit_TinyUSB.h:41:0,
                 from C:\Users\sanujadithya\Documents\Arduino\libraries\Adafruit_TinyUSB_Arduino-master\examples\WebUSB\webusb_serial\webusb_serial.ino:28:
C:\Users\sanujadithya\Documents\Arduino\libraries\Adafruit_TinyUSB_Arduino-master\src/arduino/hid/Adafruit_USBD_HID.h:42:40: error: 'HID_ITF_PROTOCOL_NONE' was not declared in this scope
                     uint8_t protocol = HID_ITF_PROTOCOL_NONE,
                                        ^~~~~~~~~~~~~~~~~~~~~
C:\Users\sanujadithya\Documents\Arduino\libraries\Adafruit_TinyUSB_Arduino-master\src/arduino/hid/Adafruit_USBD_HID.h:42:40: note: suggested alternative: 'HID_PROTOCOL_NONE'
                     uint8_t protocol = HID_ITF_PROTOCOL_NONE,
                                        ^~~~~~~~~~~~~~~~~~~~~
                                        HID_PROTOCOL_NONE
webusb_serial:31:22: error: cannot declare variable 'usb_web' to be of abstract type 'Adafruit_USBD_WebUSB'
 Adafruit_USBD_WebUSB usb_web;
                      ^~~~~~~
In file included from C:\Users\sanujadithya\Documents\Arduino\libraries\Adafruit_TinyUSB_Arduino-master\src/Adafruit_TinyUSB.h:44:0,
                 from C:\Users\sanujadithya\Documents\Arduino\libraries\Adafruit_TinyUSB_Arduino-master\examples\WebUSB\webusb_serial\webusb_serial.ino:28:
C:\Users\sanujadithya\Documents\Arduino\libraries\Adafruit_TinyUSB_Arduino-master\src/arduino/webusb/Adafruit_USBD_WebUSB.h:39:7: note:   because the following virtual functions are pure within 'Adafruit_USBD_WebUSB':
 class Adafruit_USBD_WebUSB : public Stream, public Adafruit_USBD_Interface {
       ^~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\sanujadithya\AppData\Local\Arduino15\packages\Seeeduino\hardware\samd\1.8.2\cores\arduino/TinyUSB/Adafruit_TinyUSB_ArduinoCore/Adafruit_TinyUSB_Core.h:35:0,
                 from C:\Users\sanujadithya\AppData\Local\Arduino15\packages\Seeeduino\hardware\samd\1.8.2\cores\arduino/Arduino.h:173,
                 from sketch\webusb_serial.ino.cpp:1:
C:\Users\sanujadithya\AppData\Local\Arduino15\packages\Seeeduino\hardware\samd\1.8.2\cores\arduino/TinyUSB/Adafruit_TinyUSB_ArduinoCore/Adafruit_USBD_Device.h:38:22: note: 	virtual uint16_t Adafruit_USBD_Interface::getDescriptor(uint8_t, uint8_t*, uint16_t)
     virtual uint16_t getDescriptor(uint8_t itfnum, uint8_t* buf, uint16_t bufsize) = 0;
                      ^~~~~~~~~~~~~
C:\Users\sanujadithya\Documents\Arduino\libraries\Adafruit_TinyUSB_Arduino-master\examples\WebUSB\webusb_serial\webusb_serial.ino: In function 'void setup()':
webusb_serial:57:11: error: 'TinyUSBDevice' was not declared in this scope
   while( !TinyUSBDevice.mounted() ) delay(1);
           ^~~~~~~~~~~~~
C:\Users\sanujadithya\Documents\Arduino\libraries\Adafruit_TinyUSB_Arduino-master\examples\WebUSB\webusb_serial\webusb_serial.ino:57:11: note: suggested alternative: 'USBDevice'
   while( !TinyUSBDevice.mounted() ) delay(1);
           ^~~~~~~~~~~~~
           USBDevice
exit status 1
cannot declare variable 'usb_web' to be of abstract type 'Adafruit_USBD_WebUSB'

hathach, 2021-11-24:
this board is not supported, you should try to ask on their forum.

closed 2021-11-24 by hathach

benhadad, 2024-04-27:
have a Seeeduino and following the WIkipedia from Adafruit and have same issue?