adafruit/Adafruit_TinyUSB_Arduino #141: DCD_ATTR_ENDPOINT_MAX Not correctly defined for ESP32S2 with arduino-cli

jockm, 2021-11-26:
### Operating System

Windows 10

### IDE version

arduino-cli.exe  Version: 0.20.0 Commit: 553c6375 Date: 2021-11-23T14:35:08Z

### Board

ESP32S2 Dev Module

### BSP version

2.0.1

### TinyUSB Library version

Release 1.6.0

### Sketch

Sketch: hid_generic_inout

### What happened ?

```
M:\Documents\Projects\ShootingPartner\Transmitter\RxSimulator>scripts\compile
Compiling...
In file included from C:\Users\jock\OneDrive\Documents\Arduino\libraries\Adafruit_TinyUSB_Library\src\device\dcd.h:33,
                 from C:\Users\jock\OneDrive\Documents\Arduino\libraries\Adafruit_TinyUSB_Library\src\device\usbd_control.c:33:
C:\Users\jock\OneDrive\Documents\Arduino\libraries\Adafruit_TinyUSB_Library\src\device\dcd_attr.h:158:4: warning: #warning "DCD_ATTR_ENDPOINT_MAX is not defined for this MCU, default to 8" [-Wcpp]
   #warning "DCD_ATTR_ENDPOINT_MAX is not defined for this MCU, default to 8"
    ^~~~~~~
```

### How to reproduce ?

- cd to directory with hid_generic_inout sketch
- 
`arduino-cli compile 
--fqbn esp32:esp32:esp32s2:CDCOnBoot=cdc,MSCOnBoot=default,DFUOnBoot=default,UploadMode=default,PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none`

### Debug Log

You don't give me a place for notes, so I am going to put it here.  The problem is that `DCD_ATTR_ENDPOINT_MAX` is defined in `Adafruit_TinyUSB_Arduino/src/device/dcd_attr.h` and it is looking for `CFG_TUSB_MCU` to be  `OPT_MCU_ESP32S2` to be defined, and for some reason it isn't when the arduino-cli is used.

I am not even sure who's but this is, but you should be aware of it

hathach, 2021-11-29:
this should be fixed with lib 1.7, warning is due to mis-matched version between tinyusb in IDF and one in this lib.

hathach, 2021-12-02:
fixed in 1.7.0 already

closed 2021-12-02 by hathach